pybamm-team/PyBaMM #3476: Replace `nbqa-ruff` with `ruff`

Saransh-cpp, 2023-10-26:
`ruff` can now lint jupyter notebooks!

Rjchauhan18, 2023-10-27:
Hello @Saransh-cpp 
Can you please assign this issue to me if my solution is correct ?

According to this [repo](https://github.com/astral-sh/ruff-pre-commit/blob/8a43105c597ef7067b5507451f4f1b07e5f6e26d/README.md?plain=1#L37)  which is already use in pre-commit-config.yaml file. But  then also  we need add new repo in pre-commit-config.yaml file then it is able to do changes in  jupyter notebook i have tryed it locally and we only need to remove repo which nbqa_ruff uses so that no conflict happen.

[image]

Saransh-cpp, 2023-10-27:
We don't need to add the formatter for this issue. We just have to extend the linter to jupyter notebooks. The link attached by you explains how to do that - `types_or: [python, pyi, jupyter]`.

We can then check if it is raising any weird errors and ignore them.

Rjchauhan18, 2023-10-27:
Hey , I have implemented this one `types_or: [python, pyi, jupyter].` first then i tryed to solve another [issue created by you](https://github.com/pybamm-team/PyBaMM/issues/3477) which didn't work.

- so when i implement  `types_or: [python, pyi, jupyter].`  . It automatic solve and formate the jupyter notebook

Saransh-cpp, 2023-10-27:
Yes, it isn't supposed to push changes to the repository. The changes should be pushed to a new branch of your fork manually by you.

It would be better to keep both of these issues separate as this is more about lining and the other one is more about formatting.

I'll be able to help better once you open a PR for this issue!

Rjchauhan18, 2023-10-27:
I have stop E402 to change by itself but i am still not unable stop the `;` to be removed.

agriyakhetarpal, 2023-10-28:
https://github.com/astral-sh/ruff/issues/7300 and https://github.com/astral-sh/ruff/issues/8254 should be related, we might want to hold off on this issue until some patches make it upstream

Saransh-cpp, 2023-10-28:
We can just make [useless-semicolon (E703)](https://docs.astral.sh/ruff/rules/useless-semicolon/) unfixable like @Rjchauhan18 did for E402

agriyakhetarpal, 2023-10-28:
> We can just make [useless-semicolon (E703)](https://docs.astral.sh/ruff/rules/useless-semicolon/) unfixable like @Rjchauhan18 did for E402

Oh yes, that is much better.

Rjchauhan18, 2023-10-28:
Hello @agriyakhetarpal  and @Saransh-cpp  
Thank you for helping me, I have done some changes according to your suggestion and run then locally which run fine :+1:  .
Can you please review it?

Rjchauhan18, 2023-10-30:
If this https://github.com/pybamm-team/PyBaMM/issues/3478 is the case then we don't need ruff.toml so, better to wait and then change according to pyproject.toml

```
[tool.ruff]
extend-include = ["*.ipynb"]

[tool.ruff.lint.per-file-ignores]
"**.ipynb" = ["E402", "E703"]

```

closed 2023-10-31 by Saransh-cpp