pybamm-team/PyBaMM #3477: Switch to `ruff format`

Saransh-cpp, 2023-10-26:
Ruff can now format files!

Rjchauhan18, 2023-10-27:
Hello @Saransh-cpp 
Can i make PR for this ?

As mention in this issue https://github.com/pybamm-team/PyBaMM/issues/3476 
- Refering from this [repositery](https://github.com/astral-sh/ruff-pre-commit/blob/8a43105c597ef7067b5507451f4f1b07e5f6e26d/README.md?plain=1#L68) if we add  like this
```
repos:
  - repo: https://github.com/astral-sh/ruff-pre-commit
    rev: "v0.1.1"
    hooks:
      - id: ruff
        types_or: [python, pyi, jupyter]
        args: [--fix, --ignore=E741, --exclude=__init__.py]

      - id: ruff-format
        args: [--check]

```

 **Do you want it with `--check` or without?**

Rjchauhan18, 2023-10-28:
[image]
---

Hello @Saransh-cpp 
Can you tell me that is this right or test is failing ?

---
`68 files would be reformatted, 472 files left unchanged` this is coming but nothing is change as i use `--check` in pre-commit-config.toml file. but then also 

>nox > Command pre-commit run --all-files failed with exit code 1
>nox > Session pre-commit failed.

---

[image]

Saransh-cpp, 2023-10-31:
Scientific Python's guide now has instructions for ruff format - https://learn.scientific-python.org/development/guides/style/

We should follow this initially.

PR #3656 (2023-12-24): Migrate to ruff format
# Description

Sorry for creating this huge PR just before the holidays 😬 

Fixes #3477 
Also addresses #3489

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checkl [... cut]
closed 2023-12-24 by Saransh-cpp