pybamm-team/PyBaMM #3490: Enable lazy loading of submodules

arjxn-py, 2023-10-31:
Enabling lazy loading of submodules should potentially make PyBaMM import faster.
See also: https://scientific-python.org/specs/spec-0001/ (this demands a separate issue, I guess)

_Originally posted by @agriyakhetarpal in https://github.com/pybamm-team/PyBaMM/issues/3475#issuecomment-1786894302_

prady0t, 2023-11-01:
Hey can I look into it?

agriyakhetarpal, 2023-11-01:
Hi @prady0t, thanks for helping us. Doing this would be much appreciated, but it would be better to start after #3475 gets completed. While we need `import pybamm` to be faster, the expected outcome is that there must not be any changes to the user-facing API.

prady0t, 2023-11-01:
Thanks for replying. I will be looking at other issues in the meantime.

agriyakhetarpal, 2023-11-26:
See also: https://docs.python.org/3/library/importlib.html#implementing-lazy-imports; a rather crude, bare-bones implementation for delaying imports

arjxn-py, 2023-11-27:
I can look into this now. Keeping in mind that it is not a beginner level issue and @prady0t is occupied with other one.

arjxn-py, 2024-01-12:
Hey @AbhishekChaudharii, thanks for showing your interest on this but I am currently on it and going to open up a PR in a couple of days. Feel free to look into other issues 🙂

AbhishekChaudharii, 2024-01-12:
Sure no problem :)

kratman, 2024-03-22:
@arjxn-py Can we close this now?

agriyakhetarpal, 2024-03-22:
I would say we should keep it open (other, large packages are able to make it work – maybe we can too), but yes, it is @arjxn-py's call after all

kratman, 2024-03-22:
I thought the consensus was that it was not a major bottleneck so the work/code required to do it was bigger than the problem

agriyakhetarpal, 2024-03-25:
Right, maybe let's not close it because it's still within scope but lower down the priority for now – I'll change the labels.

agriyakhetarpal, 2024-03-25:
But this is a recurring problem, isn't it? I just tested this in a fresh notebook/kernel without `sys.modules` being cached and it took ~20 seconds to import PyBaMM – so it's probably still valid. At least for organisational purposes it would be nice to keep open

agriyakhetarpal, 2024-04-06:
I just noticed that https://github.com/scientific-python/lazy_loader is now being recommended for Python versions 3.11 and later, and that too, with very specific minimum patch versions to avoid races, say 3.11.9, for example. Python 3.11 being the lowest version for us will be quite many years away at the time of writing (at least four). It sucks that I would have liked to have seen this done for PyBaMM but I don't think we have a viable choice at this time, I think we can close this and re-open later (or open a new issue, whichever way works). A better and more reliable way would be to profile PyBaMM's import and see which of the bigger modules are causing the bottleneck, and if needed then implement _minor_ breaking changes or lazy loading at the functionality level manually at the time of use of a function, so that we aren't importing a lot of those functions when one runs `import pybamm`. So I'm agreeing with @kratman here and closing this issue, please feel more than free to re-open later, @arjxn-py, or to suggest an alternative enhancement to speed up the import (whose slowness remains very much a valid problem – it's just not suited to be done for the entire public API).

closed 2024-04-06 by agriyakhetarpal