pybamm-team/PyBaMM #3495: [Bug]: Suface form effective conductivity in separator

Scottmar93, 2023-11-02:
### PyBaMM Version

23.5

### Python Version

23.5

### Describe the bug

On lines 81 and 85, of pybamm/models/submodels/electrolyte_conductivity/surface_potential_form/full_surface_form_conductivity.py
is seems that the separator porosity is used in the calculation of the effective conductivity in the separator instead of the separator transport efficiency. In the end, I think this is only an order of a few mV errors in the predicted voltage.

[image]

### Steps to Reproduce

N/A

rtimms, 2024-01-15:
@TomTranter any update on this?

TomTranter, 2024-01-15:
Completely forgot about it

PR #3905 (2024-03-19): Issue 3495 separator
# Description

Fixes #3495 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CON [... cut]
closed 2024-03-19 by tinosulzer