pybamm-team/PyBaMM #3496: Use `f-strings` everywhere for string formatting/concatenation

Saransh-cpp, 2023-11-02:
I've noticed that PyBaMM uses different (`+`, `.format`, and `f-strings`) string operations in different places. We should replace all the `+` and `.format` operations with `f-strings`. `f-strings` are more efficient (than `+`) and much more readable (than `+` and `.format`).

[[PEP 498](https://peps.python.org/pep-0498/)]

agriyakhetarpal, 2023-11-02:
Good idea, I wish there were a pre-commit hook that could do this

Saransh-cpp, 2023-11-02:
Oops, I should have mentioned it. There is 😉 

`pyupgrade` or `ruff` configured with `UP`.

I need to work on PyBaMM's ruff config.

Rjchauhan18, 2023-11-04:
><h2 id="flynt-fly" style="box-sizing: inherit; font-weight: 300; letter-spacing: -0.01em; font-size: 1.5625em; line-height: 1.4; margin: 1.6em 0px 0.64em; color: rgb(255, 255, 255); font-family: Roboto, -apple-system, BlinkMacSystemFont, Helvetica, Arial, sans-serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(38, 18, 48); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">flynt (FLY)<a class="headerlink" href="https://docs.astral.sh/ruff/rules/#flynt-fly" title="Permanent link" style="box-sizing: inherit; -webkit-tap-highlight-color: transparent; color: var(--md-default-fg-color--lighter); text-decoration: none; word-break: break-word; transition: color 0.25s ease 0s, opacity 125ms ease 0s; display: inline-block; opacity: 0; margin-left: 0.5rem;">#</a></h2><p style="box-sizing: inherit; color: rgb(255, 255, 255); font-family: Roboto, -apple-system, BlinkMacSystemFont, Helvetica, Arial, sans-serif; font-size: 17.6px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(38, 18, 48); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">For more, see<span> </span><a href="https://pypi.org/project/flynt/" style="box-sizing: inherit; -webkit-tap-highlight-color: transparent; color: var(--md-typeset-a-color); text-decoration: none; word-break: break-word; transition: color 125ms ease 0s;">flynt</a><span> </span>on PyPI.</p><div class="md-typeset__scrollwrap" style="box-sizing: inherit; margin: 1em -0.8rem; overflow-x: auto; touch-action: auto; color: rgb(255, 255, 255); font-family: Roboto, -apple-system, BlinkMacSystemFont, Helvetica, Arial, sans-serif; font-size: 17.6px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(38, 18, 48); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div class="md-typeset__table" style="box-sizing: inherit; display: inline-block; margin-bottom: 0.5em; padding: 0px 0.8rem; min-width: 100%;">

Code | Name | Message |  
-- | -- | -- | --
FLY002 | static-join-to-f-string | Consider {expression} instead of string join | 🛠️ 🧪

></div></div><br class="Apple-interchange-newline">flynt (FLY)[#](https://docs.astral.sh/ruff/rules/#flynt-fly)
>For more, see [flynt](https://pypi.org/project/flynt/) on PyPI.

Code	Name	Message	
>FLY002	[static-join-to-f-string](https://docs.astral.sh/ruff/rules/static-join-to-f-string/)	Consider {expression} instead of string join	🛠️ 🧪

[This may be useful ](https://docs.astral.sh/ruff/rules/#flynt-fly)

prady0t, 2023-11-05:
Can I work on this?

agriyakhetarpal, 2023-11-05:
Hi @prady0t, thanks for helping us. Please go ahead! As a bonus, if you can add relevant configuration for `pyupgrade` and `ruff` to do this task as @Saransh-cpp reports, that would be great.

prady0t, 2023-11-05:
Sure! Will start right ahead. Will contact you if needed. Is there a slack or discord channel btw I couldn't find one?

agriyakhetarpal, 2023-11-05:
The Slack invite is available at this link: https://pybamm.org/community/

prady0t, 2023-11-05:
Cool thanks!

prady0t, 2023-11-05:
Do I need to change uses of `.format` from docs (jupiter nootbook) as well?

agriyakhetarpal, 2023-11-06:
Yes, that wiild be great. F-strings are neater than string concatenations and `.format ()` everywhere, notebooks included!

prady0t, 2023-11-21:
Some uses of format() are bit tricky to remove. For example:
https://github.com/pybamm-team/PyBaMM/blob/6851496ff1a9cc3ec1573033cec28a2353857fbf/pybamm/solvers/jax_bdf_solver.py#L1019
Any suggestions on how to remove it. I think it's better to keep it as it is.

prady0t, 2023-11-21:
Another one! 
https://github.com/pybamm-team/PyBaMM/blob/6851496ff1a9cc3ec1573033cec28a2353857fbf/pybamm/models/submodels/active_material/total_active_material.py#L39

agriyakhetarpal, 2023-11-21:
Totally fine if such complex instances cannot be removed at this time, you can akip them or work on them later! In theory, it should be possible with every string operation. We might be able to help you better once you raise a PR.

prady0t, 2023-11-22:
I've opened a PR and made it a draft. I've converted almost all of the `format()` to `f'string'` in `.py` files. I'll keep updating this PR.

kratman, 2023-11-22:
> I've opened a PR and made it a draft. I've converted almost all of the `format()` to `f'string'` in `.py` files. I'll keep updating this PR.

I will be happy to help review when you are done

PR #3579 (2023-12-15): Using pyupgrade to convert string formatting to `f-string`.
Fixes #3496 

Linked to PR #3551
closed 2023-12-15 by Saransh-cpp