pybamm-team/PyBaMM #3506: Theoretical energy integral is very slow

tinosulzer, 2023-11-08:
This for loop is killing us and should be vectorized: https://github.com/pybamm-team/PyBaMM/blob/33d21874fbc39fd81ff89a8cc5638f6a81f8c31d/pybamm/models/full_battery_models/lithium_ion/electrode_soh.py#L1007-L1011

tinosulzer, 2023-11-08:
[image]

AbhishekChaudharii, 2023-11-09:
I tried vectorizing it but it seems that performing binary operations between NumPy arrays and `<class 'pybamm.parameters.parameter_values.ParameterValues'>`  might not be supported. 

Error :  `NotImplementedError: BinaryOperator not implemented for symbols of type <class 'numpy.ndarray'> and <class 'pybamm.expression_tree.scalar.Scalar'>`

tinosulzer, 2023-11-09:
Ok I'll look into this

PR #3510 (2023-11-13): #3506 vectorize theoretical energy
# Description

Fixes #3506 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ pre-commit run` (or `$ nox -s pre-comm [... cut]
closed 2023-11-13 by tinosulzer