pybamm-team/PyBaMM #3518: Rewrite GitHub Actions job matrices to avoid repetition

agriyakhetarpal, 2023-11-11:
The `test_on_push.yml` workflow file is slightly long (420+ lines) because there is a lot of repetition of job steps across parallel jobs. The jobs can be refactored into a single job that produces a three-dimensional matrix to run them in parallel. This way, one can write most (related jobs) with extra steps based on certain conditions and avoid repeated job configurations. The trio of unit tests + integration + coverage tests and the pair of example scripts + example notebooks checks would be sufficient for this purpose.

See https://docs.github.com/en/actions/using-jobs/using-a-matrix-for-your-jobs#example-expanding-configurations for more details. Note that it would require re-naming the jobs accordingly.

vanshiz, 2023-12-02:
I would like to work on this!

agriyakhetarpal, 2023-12-16:
Hi @vanshiz, how is this going? To explain things further – the way to go about this would be like skipping integration tests if the running job (defined in the matrix) isn't integration but some other specific test. This way, we can have  extra steps listed for a singular job (the common ones + those with specific conditions), but less jobs altogether.

vanshiz, 2023-12-16:
Noted. Working on it

agriyakhetarpal, 2024-02-07:
Actually, we don't want to do this, because CI/CD pipelines are pretty simple to debug owing to their separated nature in comparison to code files, and therefore we wouldn't mind a slightly longer YAML file (messing with `if:` conditions and boolean operators is much more prone to error). I shall close this issue for now, @vanshiz you are more than welcome to look for other issues to fix for us.

closed 2024-02-07 by agriyakhetarpal