pybamm-team/PyBaMM #3521: Accelerate cycling aging by introducing time-accelerating-factor

Brushnie33, 2023-11-14:
### Description

The cycling aging simulation is very time-consuming, for example, for 100 cycles (C/3 charge and discharge) it takes more than 2 hours, and we might not see significant amount of aging after 100 cycles. I'm wondering if it is possible to add a time-accelerating factor or other methods to speed up the process. Here is one example of how COMSOL uses a time-accelerating factor for SEI growth:
https://www.comsol.com/model/1d-lithium-ion-battery-model-for-the-capacity-fade-tutorial-12667

" By assuming that every simulated charge-discharge-cycle in the model represents an average aging behavior for a larger number of cycles Tau, and by further assuming that, over one complete charge-discharge cycle, all lithium captured in the SEI layer can be seen as stemming from the negative electrode, the capacity loss can be accelerated by rewriting the 
stoichiometry of the SEI forming reaction by adding the reaction formula:
[image]

brosaplanella, 2023-11-15:
Can't this just be implemented by updating the SEI parameters or is there something more fundamental?

tinosulzer, 2024-03-07:
Closing due to no reply from author. As @brosaplanella  said you can just change the parameters to make aging faster. It would be challenging to implement this in a way that works well for all degradation models, and a half-working implementation might just confuse users. Reopen if further discussion is needed.

closed 2024-03-07 by tinosulzer