pybamm-team/PyBaMM #3524: Pass array of times to `step`

rtimms, 2023-11-15:
When you call `solver.step` you pass in `dt`, the time time increment you want to step forward in. You can also pass `npts` to get the solution back at a linearly spaced array of times between the initial time, `t0`, and `t0+dt`. It would be good to instead be able to pass an array of times, `t_eval`, like you pass to solve, so you can get step solutions returned at non-linearly spaced times.

The motivation for this is that right now you can't pass a `starting_solution` to `Simualtion.solve` if you aren't using an experiment, so if you want to chain solutions together you have to call `Simulation.step` instead. 

An alternative is that we allow starting solutions without experiments in `Simulation.solve`.

The whole logic for `step` and `solve` is pretty clunky both in `Simulation` and `BaseSolver`, but that's a separate issue...

prady0t, 2023-11-25:
Can you please give examples of how to use step() function to get solutions which I can use for debugging? I'm quite new here and I cant find it in project doc.

rtimms, 2023-12-07:
Sorry for the slow reply. Take a look [here](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/scripts/SPMe_step.py).

prady0t, 2023-12-16:
Maybe if we change step function to something like: 
```py
    def step(
        self,
        old_solution,
        model,
        dt=0,
        non_linear_time=None,
        npts=2,
        inputs=None,
        save=True,
    )
```
Here `non_linear_time` is an array of time (example: [0,10,15,17,20]). We can assign `dt = non_linear_time[i] - non_linear_time[i-1]` inside a loop and make changes to the function accordingly. Is this a correct solution?

arjxn-py, 2023-12-16:
> Here `non_linear_time` is an array of time (example: [0,10,15,17,20]).

Hi @prady0t, theoretically it looks promising, though it would be nicer to open a PR to keep things moving & also would be easier for us to suggest changes there 🙂

prady0t, 2023-12-16:
Sure will do!

prady0t, 2023-12-17:
> > Here `non_linear_time` is an array of time (example: [0,10,15,17,20]).
> 
> Hi @prady0t, theoretically it looks promising, though it would be nicer to open a PR to keep things moving & also would be easier for us to suggest changes there 🙂

Made a PR. Please have a look.

PR #3627 (2024-03-18): Implemented `non_linear_time`, an array of time as an input to `step` function
# Description

Fixes #3524 

Now  an array of time can be passed to `step` function which will return solutions at that step times.
Example:

```py
step_solver.step(step_solution, model, non_linear_time=[0, 1000, 1500, 1700, 2000])
```
Output:

[image]

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes [... cut]
closed 2024-03-18 by kratman