pybamm-team/PyBaMM #3528: Issue with IDAKLU installation from source

brosaplanella, 2023-11-16:
I am doing a fresh install on Python 3.11 (in WSL). The comman `nox -s pybamm-requires` worked fine, but when running `nox -s dev` I got the error below, related to IDAKLU.

Has anyone else encountered this?

```bash
      copying pybamm/../CMakeBuild.py -> build/lib.linux-x86_64-cpython-311/pybamm/..
      running build_ext
      ---------- Running CMake for idaklu solver ----------------------------------------
      -- The C compiler identification is GNU 9.4.0
      -- The CXX compiler identification is GNU 9.4.0
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: /usr/bin/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- Detecting CXX compiler ABI info
      -- Detecting CXX compiler ABI info - done
      -- Check for working CXX compiler: /usr/bin/c++ - skipped
      -- Detecting CXX compile features
      -- Detecting CXX compile features - done
      -- pybind11 v2.12.0 dev1
      CMake Warning (dev) at pybind11/tools/FindPythonLibsNew.cmake:101 (message):
        Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
        are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
        the cmake_policy command to set the policy and suppress this warning, or
        preferably upgrade to using FindPython, either by calling it explicitly
        before pybind11, or by setting PYBIND11_FINDPYTHON ON before pybind11.
      Call Stack (most recent call first):
        pybind11/tools/pybind11Tools.cmake:50 (find_package)
        pybind11/tools/pybind11Common.cmake:188 (include)
        pybind11/CMakeLists.txt:222 (include)
      This warning is for project developers.  Use -Wno-dev to suppress it.

      -- Found PythonInterp: /home/brosaplanella/PyBaMM/venv/bin/python (found suitable version "3.11.6", minimum required is "3.6")
      -- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.11.so
      -- Performing Test HAS_FLTO
      -- Performing Test HAS_FLTO - Success
      Traceback (most recent call last):
        File "<string>", line 1, in <module>
      ModuleNotFoundError: No module named 'casadi'
      Found python casadi path:
      Trying to link against python casadi package
      CMake Error at CMakeLists.txt:80 (find_package):
        Could not find a package configuration file provided by "casadi" with any
        of the following names:

          casadiConfig.cmake
          casadi-config.cmake

        Add the installation prefix of "casadi" to CMAKE_PREFIX_PATH or set
        "casadi_DIR" to a directory containing one of the above files.  If "casadi"
        provides a separate development package or SDK, be sure it has been
        installed.

      -- Configuring incomplete, errors occurred!
      ---------- Building idaklu module ----------------------------------------
      make: Makefile: No such file or directory
      make: *** No rule to make target 'Makefile'.  Stop.
      error: can't copy '/home/brosaplanella/PyBaMM/build/temp.linux-x86_64-cpython-311/idaklu.cpython-311-x86_64-linux-gnu.so': doesn't exist or not a regular file
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pybamm
  Running setup.py clean for pybamm
Failed to build pybamm
ERROR: Could not build wheels for pybamm, which is required to install pyproject.toml-based projects
nox > Command /home/brosaplanella/PyBaMM/venv/bin/python -m pip install '.[all,dev,jax,odes]' failed with exit code 1
nox > Session dev failed.
```

agriyakhetarpal, 2023-11-16:
I have encountered this issue earlier (a lot, actually! 😃), but it should be nothing to worry. Could you uninstall or `--force-reinstall` `casadi` and try installing PyBaMM with updated `setuptools` and `wheel`?

brosaplanella, 2023-11-16:
Is that with the venv activated or deactivated?

agriyakhetarpal, 2023-11-16:
Deactivated, since the `nox -s dev` session now creates a folder `venv/` for you and installs all the dependencies inside it

brosaplanella, 2023-11-16:
Converting it into a discussion as it might be useful to other people.