pybamm-team/PyBaMM #3533: Add a workflow to drop old versions of core dependencies

agriyakhetarpal, 2023-11-17:
We could add a workflow to open an issue once every quarter about examining and dropping support for versions of core dependencies that are nearing the end of their support window (and bump the versions manually)

https://scientific-python.org/specs/spec-0000/

_Originally posted by @agriyakhetarpal in https://github.com/pybamm-team/PyBaMM/pull/3501#discussion_r1396867756_

Saransh-cpp, 2023-11-25:
I've looked into multiple scientific libraries and I think it should be okay to not strictly follow SPEC 0000. The Python version should only be dropped when it reaches its EOL or if it is blocking the usage of a new feature. For example, Python 3.7 blocked the usage of new type hints -

```py
# Python 3.7
from typing import List, Union
a: List[Union[str, int]] = ["Hello", 1]

# Python 3.8+
from __future__ import annotations
a: list[str | int] = ["Hello", 1]

# Python 3.11
a: list[str | int] = ["Hello", 1]
```

Also see https://scikit-hep.org/supported-python-versions.

I think we can close this issue and bump versions manually whenenever we run into problems in the future.

agriyakhetarpal, 2023-11-25:
Thanks for looking into this and the information, @Saransh-cpp! Closing this. I wish that Scikit-HEP and Scientific Python are able to bring each other to a consensus via a unified version bounds policy someday...

closed 2023-11-25 by agriyakhetarpal