pybamm-team/PyBaMM #3543: [Bug]: Thermal model not working for a composite electrode particle model with two (or more) phases

chmabaur, 2023-11-20:
### PyBaMM Version

23.5

### Python Version

3.9.13

### Describe the bug

Lumped thermal model is not compatible with the composite particle phase option.

### Steps to Reproduce

```
import pybamm

pybamm.set_logging_level("INFO")
# load models
options = {
    "particle phases": ("2", "1"),
    "open-circuit potential": (("single", "current sigmoid"), "single"),
    "cell geometry": "arbitrary", "thermal": "lumped"
}
model = pybamm.lithium_ion.DFN(options)

parameter_values = pybamm.ParameterValues("Chen2020_composite")

# create and run simulations
sim = pybamm.Simulation(model, parameter_values=parameter_values)
sim.solve([0, 4500])
```

### Relevant log output

```shell
File ~\AppData\Local\Programs\Python\Environments\gen-env\lib\site-packages\pybamm\models\full_battery_models\lithium_ion\dfn.py:32 in __init__
    self.set_submodels(build)

  File ~\AppData\Local\Programs\Python\Environments\gen-env\lib\site-packages\pybamm\models\full_battery_models\lithium_ion\base_lithium_ion_model.py:59 in set_submodels
    self.build_model()

  File ~\AppData\Local\Programs\Python\Environments\gen-env\lib\site-packages\pybamm\models\full_battery_models\base_battery_model.py:907 in build_model
    self._build_model()

  File ~\AppData\Local\Programs\Python\Environments\gen-env\lib\site-packages\pybamm\models\base_model.py:550 in _build_model
    self.build_coupled_variables()

  File ~\AppData\Local\Programs\Python\Environments\gen-env\lib\site-packages\pybamm\models\base_model.py:480 in build_coupled_variables
    raise pybamm.ModelError(

ModelError: Missing variable for submodel 'thermal': 'Negative electrode reaction overpotential [V]'.
Check the selected submodels provide all of the required variables.
```

chmabaur, 2023-11-29:
Isn't this a relatively simple bug to fix in the BaseThermal class? Something along this line for the heating term due to the reaction overpotential without having gone through the math. Something similar is then also required for the entropic heating term, but that is all no?

```
a_j_p = variables["Positive electrode volumetric interfacial current density [A.m-3]"]

if int(self.options['particle phases'][1]) > 1:
    a_j_p_p = variables["Positive electrode primary volumetric interfacial current density [A.m-3]"]
    a_j_p_s = variables["Positive electrode secondary volumetric interfacial current density [A.m-3]"]
    eta_r_p_p = variables["Positive electrode primary reaction overpotential [V]"]
    eta_r_p_s = variables["Positive electrode secondary reaction overpotential [V]"]
    
    Q_rxn_p = a_j_p_p * eta_r_p_p + a_j_p_s * eta_r_p_s
else:
    eta_r_p = variables["Positive electrode reaction overpotential [V]"]
    Q_rxn_p = a_j_p * eta_r_p
```

rtimms, 2023-12-01:
Yes, I'd implement as a sum over phases e.g. like [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/submodels/interface/total_interfacial_current.py#L45).

Equations, summing over phases, m

[image]

PR #3586 (2023-12-05): Fix Issue 3543 - Thermal model not working for a composite electrode particle model with two (or more) phases
# Description

Summing the irreversible and reversible heating terms over the phases in the BaseThermal class as detailed in https://github.com/pybamm-team/PyBaMM/issues/3543#issuecomment-1836406296. To do so, the entropic change dUdT variables are stored for each phase in the BaseOpenCircuitPotential class.

Fixes #3543

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change [... cut]
closed 2023-12-05 by tinosulzer