pybamm-team/PyBaMM #3552: [Bug]: Half-cell, particle distribution DFN fails during charge experiments

Sergio-PY, 2023-11-23:
### PyBaMM Version

23.5

### Python Version

3.9.18

### Describe the bug

Hello,

When i create a half-cell DFN model and I try to include the particle distribution for the positive electrode, it wouldn't take any charge procedure. It only discharges, but the charge always fails. 
When i do exactly the same but without including the particle distribution everything works fine.

### Steps to Reproduce

`import pybamm`

Create the model using the particle size distribution option as well as the half cell configuration.

`options = {"particle size": "distribution","working electrode": "positive" }`
`MPDFN_model = pybamm.lithium_ion.DFN(options=options)`

Import half cell NMC parameters

`param_nmc = pybamm.ParameterValues("Xu2019")`

Create size distribution for the positive particles

`param_nmc_MP=pybamm.get_size_distribution_parameters(param_nmc, electrode="positive")`

Load a simple experiment

`experiment = pybamm.Experiment(["Discharge at C/25 until 3.5 V", "Charge at C/25 until 4.2 V"])`

Load everything (regardless of the solver used the result and error is the same)

`MP_sim_pOCV = pybamm.Simulation(MPDFN_model, parameter_values=param_nmc_MP, solver=pybamm.CasadiSolver(atol=1e-3, rtol=1e-3, mode="safe"), experiment=experiment)`

Simulate. The error always arise in the next step.

`MP_sol_pOCV = MP_sim_pOCV.solve()`

The error always gives a problem when charging, when i just discharge is ok.

### Relevant log output

```shell
---------------------------------------------------------------------------
NotImplementedError                       Traceback (most recent call last)
Cell In[7], line 3
      1 # MP-DFN
      2 MP_sim_pOCV = pybamm.Simulation(MPDFN_model, parameter_values=param_nmc_MP, solver=pybamm.CasadiSolver(atol=1e-3, rtol=1e-3, mode="safe"), experiment=experiment)
----> 3 MP_sol_pOCV = MP_sim_pOCV.solve()

File ~\.conda\envs\PyBaMM_env\lib\site-packages\pybamm\simulation.py:725, in Simulation.solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, callbacks, showprogress, **kwargs)
    723 npts = max(int(round(dt / op_conds.period)) + 1, 2)
    724 try:
--> 725     step_solution = solver.step(
    726         current_solution,
    727         model,
    728         dt,
    729         npts=npts,
    730         save=False,
    731         **kwargs,
    732     )
    733 except pybamm.SolverError as error:
    734     if (
    735         "non-positive at initial conditions" in error.message
    736         and "[experiment]" in error.message
    737     ):

File ~\.conda\envs\PyBaMM_env\lib\site-packages\pybamm\solvers\base_solver.py:1165, in BaseSolver.step(self, old_solution, model, dt, npts, inputs, save)
   1163         model.y0 = old_solution.all_ys[-1][:, -1]
   1164     else:
-> 1165         _, concatenated_initial_conditions = model.set_initial_conditions_from(
   1166             old_solution, return_type="ics"
   1167         )
   1168         model.y0 = concatenated_initial_conditions.evaluate(
   1169             0, inputs=model_inputs
   1170         )
   1172 set_up_time = timer.time()

File ~\.conda\envs\PyBaMM_env\lib\site-packages\pybamm\models\base_model.py:576, in BaseModel.set_initial_conditions_from(self, solution, inplace, return_type)
    574 if isinstance(var, pybamm.Variable):
    575     try:
--> 576         final_state = solution[var.name]
    577     except KeyError as e:
    578         raise pybamm.ModelError(
    579             "To update a model from a solution, each variable in "
    580             "model.initial_conditions must appear in the solution with "
    581             "the same key as the variable name. In the solution provided, "
    582             f"'{e.args[0]}' was not found."
    583         )

File ~\.conda\envs\PyBaMM_env\lib\site-packages\pybamm\solvers\solution.py:510, in Solution.__getitem__(self, key)
    507     return self._variables[key]
    508 else:
    509     # otherwise create it, save it and then return it
--> 510     self.update(key)
    511     return self._variables[key]

File ~\.conda\envs\PyBaMM_env\lib\site-packages\pybamm\solvers\solution.py:470, in Solution.update(self, variables)
    468         model._variables_casadi[key] = var_casadi
    469     vars_casadi.append(var_casadi)
--> 470 var = pybamm.ProcessedVariable(
    471     vars_pybamm, vars_casadi, self, cumtrapz_ic=cumtrapz_ic
    472 )
    474 # Save variable and data
    475 self._variables[key] = var

File ~\.conda\envs\PyBaMM_env\lib\site-packages\pybamm\solvers\processed_variable.py:105, in ProcessedVariable.__init__(self, base_variables, base_variables_casadi, solution, warn, cumtrapz_ic)
    102     self.initialise_2D()
    103 else:
    104     # Raise error for 3D variable
--> 105     raise NotImplementedError(
    106         "Shape not recognized for {} ".format(base_variables[0])
    107         + "(note processing of 3D variables is not yet implemented)"
    108     )

NotImplementedError: Shape not recognized for 48230.0 * y[0:12000] (note processing of 3D variables is not yet implemented)
```

Sergio-PY, 2023-11-24:
Ok, i have seen that the error is general for all the particle distribution implementations. No DFN + particle distribution works when it has to do a charge process. I imagine that is a well-known issue and there are probably people working on that.

rtimms, 2023-11-27:
Thanks for raising. The issue is with post-processing of 3D variables. Duplicate of https://github.com/pybamm-team/PyBaMM/issues/1549

closed 2023-11-27 by rtimms