pybamm-team/PyBaMM #3564: Migrate to `scikit-build-core` or `meson-python`

agriyakhetarpal, 2023-11-27:
Now that we have a declarative format for build metadata provided by `pyproject.toml` and build isolation enabling a more stable installation procedure via PEP 517/518, it eases the migration to either the [`scikit-build-core`](https://scikit-build-core.readthedocs.io/en/latest/index.html) or the [`meson-python`](https://meson-python.readthedocs.io/en/latest/) build-backend(s). Both of them include support for editable installations and are slowly becoming more mature for downstream users, considering that NumPy[^1] and SciPy[^2] have successfully transitioned this year.

The `scikit-build-core` build backend would still use CMake as a build system as we do currently, but `meson-python` requires [`ninja`](https://ninja-build.org/) as a build system, which would render the IDAKLU target compilation procedure radically different from the current one.

Some advantages of these new build-backends:

1. Better support for compilers plus ease of use[^3]
2. Reliable build caching for faster installation
3. Both of them are faster at compilation and more modern (multithreaded builds, cross-compilation, do not use `distutils` or `setuptools` at all)
4. Packagers for libraries in the Scientific Python ecosystem are slowly migrating and we should too
5. Might enable faster IDAKLU compilation with other compilers like MinGW? i.e., on Windows when building from source (which we do only when building the wheel)

[^1]: https://rgoswami.me/posts/numpy-meson-f2py/
[^2]: https://labs.quansight.org/blog/building-scipy-with-flang
[^3]: https://labs.quansight.org/blog/2021/07/moving-scipy-to-meson

kratman, 2023-11-27:
Personally I would prefer to work with CMake over Ninja, but that could just be my bias from spending most of my career as a C++ dev

cringeyburger, 2024-02-29:
Hello! I was going through the GSOC 2024 Idealist and encountered this project. I am interested in learning and working on the Python backend, and this project will significantly help develop my skills. In the same route, I can help contribute to PyBaMM!

As a starter, I would like to discuss whether we should decide on `meson-python` or `scikit-build-core`.

I would prefer `meson-python` over `scikit-build-core`, mainly because of its popularity after SciPy (version 1.9.0) and [NumPy](https://numpy.org/doc/stable/reference/distutils_status_migration.html) (version 1.26) decided to migrate to it. The documentation, posts, blogs and projects that sparked because of this would be helpful. 

Also, `meson-python` is the [most popular](https://news.ycombinator.com/item?id=39006295 ) method after `setuptools` to build C extensions. 

I would also like to discuss some technical aspects of the project. Thank you!

agriyakhetarpal, 2024-02-29:
Thanks for sharing, @cringeyburger. I would agree that `meson-python` is being used highly, but NumPy and SciPy use a vendored version of it because they want features that do not yet exist in the main source code. The reason, I believe, is that the Meson developers tend to prevent pushing anything hacky into the codebase and wish to do things in the conventional way and how robust solutions go. Both projects are very large and use almost every feature available in Meson, which we may not require for our use case – that should be great.

The Ninja build system should be able to handle the IDAKLU solver's compilation, but I am **unsure** whether it can do so for the prerequisites (SuiteSparse and SUNDIALS) – they might require a pure CMake build or perhaps calling CMake from inside `meson.build`. I am aware that `pybind11` works with `meson-python` and can be used as normal, so it should just be a matter of pointing the build procedure to the location of the compiled files for the prerequisites. I would perhaps explore methods for caching so that editable installations can be sped up – Meson's in-place builds on the `import pybamm` statement can then rely on just compiling specific sections of the code – my experiments with https://github.com/pybop-team/PyBOP/pull/176 revealed that around 30 seconds are consumed by the IDAKLU compilation, and ~1 min 30 seconds for SuiteSparse and SUNDIALS on Unix-based systems. 

However, Windows builds are significantly long and take ~20 minutes overall, without caching which would be a problem on editable installations. It looks like https://github.com/microsoft/vcpkg/pull/36885 will get merged soon which could provide some benefits (I'm unsure whether it is the SUNDIALS compilation or the SuiteSparse's component's compilation that takes longer on Windows). This might require bumping SUNDIALS to v6.7.0, we use v6.5.0 – it doesn't look like there are many changes to the IDAS component that we use (here are the [release notes](https://github.com/LLNL/sundials/releases/tag/v6.7.0)) and subsequently perhaps SuiteSparse from v6.0.3 to a later version (which would require changes to the C++ solver code).

We should be able to forego `setup.py` entirely with Meson – it has been done for NumPy: https://github.com/numpy/numpy/pull/24519 and should be out with NumPy v2 (any time now). With `scikit-build-core`, we might not be able to do so completely, but it isn't an issue, we can remove the dependence on `wheel.bdist_wheel` (it has been recommended against using it for quite some time now) and rewrite the `setup.py` code to be cleaner. Like it has been mentioned above, using this choice of backend will be much easier to integrate – everything will be governed by the CMake flags and `CMakeLists.txt`, which we already have.

agriyakhetarpal, 2024-02-29:
Bumping the priority to medium since this will now be worked on this year

agriyakhetarpal, 2024-02-29:
I did a quick test and I notice that on macOS, `vcpkg install sundials` takes 14 seconds, while `vcpkg install suitesparse` took 5.5 minutes, so perhaps using `sundials[klu]` when it is ready upstream will also resolve our Windows woes? Building on Windows is slow in general and our Windows builds painfully take 20 or so minutes and this is after enabling multithreaded builds (without which it takes even longer). But I see that SuiteSparse uses MinGW to compile in Windows CI where the time is around 2 minutes; we compile from source using GCC on Unix-based platforms where this hardly takes 30 seconds – I assume adding `Ccache` support will be highly beneficial for Windows

cringeyburger, 2024-02-29:
On windows x64 architecture laptop, `vcpkg install sundials` took 19 seconds and `vcpks install suitesparse` took a solid 16 minutes
- I forgot that I had `ninja VER 1.11.0` installed, so it defaulted to it
- For Fortran, I had nothing installed before hand so it used `MinGW Fortran`
- Though the time it took to handle `suitesparse:x64-windows` is 1.8 minutes.

As you have said, if that PR with vcpkg is merged, then having `suitesparse[klu]` can help speed up the process.

I am interested in learning more about `Ccache`, but from what I know, caching will definitely help improve the successive build processes.

For large packages like ours, the incremental build feature of `CMake` utilised by `scikit-build-core` would work well with `Ccache`.

Assuming that majority of our team is familiar with `CMake`, I too think that scikit-build-core is the better option.

agriyakhetarpal, 2024-03-01:
2.8 minutes should be reasonable, thanks – this confirms that the SuiteSparse installation is the slowest installation out of all. 2.8 minutes is of course significantly slower than installing CasADi from its Python wheel, which is a matter of seconds – we were using the wheel earlier but had to resort to using the registry afterwards due to upstream changes. We have an issue open (#3603) for restoring this behaviour, which no one has been able to get to yet.

It looks like the SuiteSparse issue will be dormant for a long time – I noticed that I had seen it before because I was already subscribed to it :)

The CasADi installation from `vcpkg` looks normal to me especially if it succeeded, since I have noted these logs in our Windows wheels before – but it would be good to look how to clear out the warnings (they do not look to be relevant for the CasADi installation and are probably more so for the registry and its portfile itself).

cringeyburger, 2024-03-02:
I tried fixing the warnings by changing the portfile. I followed the instructions given through warnings when I built it.
1. Added `vcpkg-cmake-config` as a dependency and added `file(REMOVE_RECURSE "${CURRENT_PACKAGES_DIR}/debug/include")`. This helped in reducing one warning.
2. I tried to fix warnings 2 and 3 using `vcpkg_cmake_config_fixup()`, but it's erroring out because it's searching for some directory that doesn't exist. I tried with and without this command, and the directory `CONFIG_PATH share/casdi/cmake` doesn't exist.
3. Regarding number 4, a quick search through the net tells me we may have to fix the absolute paths in the header files themselves.

agriyakhetarpal, 2024-03-02:
Oh, by "good to look how to clear the warnings" I didn't mean doing so just now – I do not think it is the biggest of deals currently and can definitely be picked up at a later time (maybe CasADi can be added as a `vcpkg` package upstream too where someone interested can take care of this). Nevertheless, thanks for attempting this!

The gist of the discussion that entailed here should be that despite snd regardless of the build backend we end up choosing, reducing the Windows installation time would be great. We don't compile the IDAKLU solver on PR workflows because of that – that is an area where reducing CI time is important. But then we don't compile it when running the Windows scheduled tests either (#3664) where we don't need to bother how long the test runs. Being able to speed things up would benefit us in other ways too, say, identifying issues such as #3783 early on.

prady0t, 2024-03-12:
I'm interested in this project too. I'm in the process of making a proposal

PR #4487 (2025-01-07): Download IDAKLU from pybammsolvers
# Description

This will separate the IDAKLU C++ code from pybamm.

Fixes #3564
Fixes #4611

This ticket is something that can be handled in the other repo
Closes #3603 

## Type of change

This should speed up CI by skipping the build of the C++ code.

- [x] Optimization (back-end change that speeds up the code)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [CONTRIBUTING.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CONTRIBUTING.md#installing-and-using-pre-commit) for how to set this up to run automatic [... cut]
closed 2025-01-07 by kratman