pybamm-team/PyBaMM #3566: [Bug]: Default SPMe and DFN models/examples have no NE potential or overpotential

tommaull, 2023-11-27:
### PyBaMM Version

lastest Dev and Main as of 27/11/23

### Python Version

3.9

### Describe the bug

[image]

NE response is implausible - hovering around 0 potential, with no overpotential behaviours. 

### Steps to Reproduce

Run 23.9 Main, SPMe example. Observe NE behaviour.

rtimms, 2023-11-27:
This is because we measure potential wrt the solid phase negative electrode potential at x=0, whereas I'm guessing what you're expecting is the potential measured wrt to a different reference, such as the electrolyte potential?

rtimms, 2023-11-27:
See also https://github.com/pybamm-team/PyBaMM/discussions/3470 and https://github.com/pybamm-team/PyBaMM/discussions/3254

closed 2023-11-27 by tommaull