pybamm-team/PyBaMM #3611: [Bug]: Lumped thermal model conflates cell volume with electrode volume

ejfdickinson, 2023-12-12:
### PyBaMM Version

23.9

### Python Version

3.9.13

### Describe the bug

The lumped thermal model (class `Lumped`) utilises the `"Volume-averaged total heating [W.m-3]"` variable for the cell heat equation, in conjunction with cell volume `"Cell volume [m3]"`.

However, the volume-averaged total heating variable does not respect the cell volume when computed. Instead it uses a successive averaging approach which causes the resulting value to reflect only the apparent volume of the electrodes (as electrode pairs with separator and current collector), as the product of total electrode `thickness * height * width`. Consequently, **the resulting applied total heat source on the cell is in error by the ratio of cell volume to electrode volume**.

A safer computation route for total cell heating would be the product `height * width` with the integrated heat source per unit simulated electrode-pair area (unit: W/m2).

Spotters: Darryl Doyle (@darryl-ad ), Ivan Korotkin

### Steps to Reproduce

Although the specified cell volume exceeds the product of modelled 1D thickness with electrode area (as expected due to inactive dead space in the cell), the lumped thermal model utilises a volumetric heat source (vs cell volume) that is identical to the X-averaged volumetric heat source at 1D simulation scale.

```
import pybamm

model = pybamm.lithium_ion.DFN(
    options={
        "thermal": "lumped"
    }
)

experiment = pybamm.Experiment(
    [
        "Rest for 1 s",
        "Discharge at 1C until 3.6 V"
    ]
)

parameter_values = pybamm.ParameterValues("ORegan2022")
parameter_values["Cell volume [m3]"] = 70e-3 * ((21e-3/2) ** 2) * 3.14159

sim = pybamm.Simulation(
    model=model,
    experiment=experiment,
    parameter_values=parameter_values
)

sol = sim.solve()

contributing_layers = [
    "Negative current collector",
    "Negative electrode",
    "Separator",
    "Positive electrode",
    "Positive current collector"
]

thickness = sum([
    parameter_values[layer + " thickness [m]"] for layer in contributing_layers
])
electrode_volume = thickness * parameter_values["Electrode height [m]"] * parameter_values["Electrode width [m]"]

print("Difference of local and global volumetric heat source:", sol["X-averaged total heating [W.m-3]"].entries - sol["Volume-averaged total heating [W.m-3]"].entries)
print("Global (cell) volume:", parameter_values["Cell volume [m3]"])
print("Local (modelled) volume:", electrode_volume)
```

ejfdickinson, 2023-12-14:
On further thinking, it _could_ be argued that this is really a bug in `create_from_bpx`, which ought to correct for the difference between `Cell volume [m3]` and electrode volume before applying the cell density and specific heat capacities uniformly to the electrodes.

However, the appearance of the true `cell_volume` in the heat transfer coefficient term in `Lumped` still suggests that it would really be better if `Lumped` addressed the total cell volume in all cases, not just the implied volume of electrochemically simulated active material.

rtimms, 2023-12-18:
Thanks for pointing this out. I think we should add variables for "integrated heat source per unit simulated electrode-pair area (unit: W/m2)" and "... heating [W]" as part of doing this, and make distinctions between "Volume-averaged" i.e. _the actual cell volume_ and "Apparent volume-averaged" over the "actually battery stuff". What are your suggestions for good, informative names for these?

closed 2024-01-16 by kratman