pybamm-team/PyBaMM #3613: [Bug]: SEI ('electron-migration limited') model

kawaMANMI, 2023-12-12:
### PyBaMM Version

23.9

### Python Version

3.10

### Describe the bug

I noticed that the SEI thickness decreases when the 'electron-migration limited' model (option) is used, which is unrealistic.

### Steps to Reproduce

import pybamm
import numpy as np
import matplotlib.pyplot as plt
exp_CCCV = pybamm.Experiment(
    [("Charge at C/10 until 4.2 V",
      "Rest for 10 hours",
      "Discharge at C/10 until 2.5 V",
      "Rest for 10 hours",
      )]
)
model = pybamm.lithium_ion.SPM(options={
    "SEI": "electron-migration limited",
    "SEI porosity change": "true",
},
    name="SPM-SEI")

solver = pybamm.CasadiSolver()
sim = pybamm.Simulation(model, experiment=exp_CCCV, solver=solver)
sol = sim.solve(initial_soc=0)
plt.plot(sol["Time [h]"].entries,
         sol['X-averaged negative total SEI thickness [m]'].entries, label='SEI thickness')

kawaMANMI, 2024-01-16:
Just to provide more clarity, I have attached three figures here for the electro-migration limited model before correction, after correction, and the interstitial-diffusion limited model.  The corrected version follows a similar trend to the interstitial model. 
[image]
[image]
[image]

brosaplanella, 2024-01-18:
For clarity, the current model is the bottom plot and Kawa's fix is the middle one.

PR #3622 (2024-01-18): Fixed a bug where the SEI thickness decreased at some intervals when using the 'electron-migration limited' model.
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

The SEI thickness decreased at some intervals when the 'electron-migration limited' model was used. It has been corrected by considering only negative contributions to eta_sei in the j_sei, using the Heaviside function.

Fixes #3613 

## Type of change

Please add a line in the relevant sectio [... cut]
closed 2024-01-18 by kratman