pybamm-team/PyBaMM #3623: Better error or handling when passing a str to Experiment

rtimms, 2023-12-15:
`Experiment` expects `operating_conditions` to be a list of strings. If you just pass a string, e.g. `pybamm.Experiment("Discharge at 1C until 2.5V")` you get the error `ValueError: Operating conditions must contain keyword 'for' or 'until'.`. 

We should add a check to see if `operating_conditions` is list of strings or a str. If it's a str we should convert to a list, if not we should raise an error (see e.g. `check_input_validity` for `QuickPlot` [here](https://github.com/pybamm-team/PyBaMM/blob/903323efca7fef8f630924d114ce70dcf8a92363/pybamm/plotting/quick_plot.py#L250).

kratman, 2023-12-15:
Do we need to support both strings and lists? I see no reason why we can't require the user to always send a list(str), and just add a type hint.

rtimms, 2023-12-15:
Yeah, happy with that. There are lots of places in PyBaMM where we aren't checking types. There is work to address this in some places, e.g. https://github.com/pybamm-team/PyBaMM/issues/3497

prady0t, 2023-12-16:
Can I take this up?

agriyakhetarpal, 2023-12-16:
Hi @prady0t, I see that you're assigned to #3524, how is that coming along?

prady0t, 2023-12-16:
Ah yes I'm working on it. I need a bit of assistance though (I wrote a comment about it).

prady0t, 2023-12-16:
I can work on two issues at once no big deal. 😸

kratman, 2023-12-16:
We try to only assign new developers to 1 issue at a time. If you do end up working on this issue though, you should try the type hint approach. It would make it only a couple lines.

prady0t, 2023-12-16:
Okay that's totally fine 👍🏼

PR #3650 (2024-01-03): Improved error handling and added type hinting in Experiment
# Description

1. Improved error handling by raising TypeError when a str is passed to Experiment instead of list of strings
2. Added type hinting for better readability.

Fixes #3623

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization  [... cut]
closed 2024-01-03 by kratman