pybamm-team/PyBaMM #3630: [Bug]: 23.9 - ShapeError from electrolyte equation when specifying "Ambient temperature [K]" as pybamm.Interpolant

ejfdickinson, 2023-12-18:
### PyBaMM Version

23.9

### Python Version

3.9.13

### Describe the bug

As of 23.9, using a `pybamm.Interpolant` to specify the `"Ambient temperature [K]"`, to run a spatially isothermal model with a specified transient temperature profile, raises a `ShapeError` within `Discretisation.process_dict()` acting on the equation with key `"Electrolyte current density [A.m-2]"`.

This parameter definition pattern was supported and functional in 23.5.

### Steps to Reproduce

Works in 23.5. Fails in 23.9.

```
import pybamm
import numpy as np
import matplotlib.pyplot as plt

times = np.arange(0,1810,10)
tmax = max(times)
experiment = pybamm.Experiment([f"Discharge at 1C for {tmax} s"])

temp_drive_cycle = pybamm.Interpolant(
    times,
    298.15 + 20 * (times / tmax),
    pybamm.t
)

parameter_values = pybamm.ParameterValues("Chen2020")
parameter_values.update(
    {
        "Initial temperature [K]": 298.15,
        "Ambient temperature [K]": temp_drive_cycle
    }
)

sim = pybamm.Simulation(
    model=pybamm.lithium_ion.DFN(),
    parameter_values=parameter_values,
    experiment=experiment
)

sol = sim.solve()
sol.plot()

plt.plot(sol["Time [s]"].entries, sol["Volume-averaged cell temperature [K]"].entries - 273.15)
plt.xlabel("Time / s")
plt.ylabel("Cell temperature / degC");
```

### Relevant log output

```shell
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
File ...\pybamm\discretisations\discretisation.py:728, in Discretisation.process_symbol(self, symbol)
    727 try:
--> 728     return self._discretised_symbols[symbol]
    729 except KeyError:

KeyError: Multiplication(0x31886afc2ee59eaf, *, children=['(concatenation(broadcast(1.6212500000000003e-11), broadcast(4.1791384878584733e-11), broadcast(2.5148216275389202e-11)) * ((maximum(concatenation(broadcast(4.0), broadcast(2.127659574468085), broadcast(2.9850746268656714)) * Porosity times concentration [mol.m-3](Negative electrode porosity times concentration [mol.m-3], Separator porosity times concentration [mol.m-3], Positive electrode porosity times concentration [mol.m-3]), 10.0)) ** 3.0)) - (concatenation(broadcast(9.92164615877829e-06), broadcast(2.557528655362438e-05), broadcast(1.5390081937322494e-05)) * ((maximum(concatenation(broadcast(4.0), broadcast(2.127659574468085), broadcast(2.9850746268656714)) * Porosity times concentration [mol.m-3](Negative electrode porosity times concentration [mol.m-3], Separator porosity times concentration [mol.m-3], Positive electrode porosity times concentration [mol.m-3]), 10.0)) ** 1.5)) + concatenation(broadcast(0.00041612500000000005), broadcast(0.0010726562857425487), broadcast(0.0006454773475772603)) * (maximum(concatenation(broadcast(4.0), broadcast(2.127659574468085), broadcast(2.9850746268656714)) * Porosity times concentration [mol.m-3](Negative electrode porosity times concentration [mol.m-3], Separator porosity times concentration [mol.m-3], Positive electrode porosity times concentration [mol.m-3]), 10.0))', '((0.00012763994028092067 * ing_functio(time) / (maximum(concatenation(broadcast(4.0), broadcast(2.127659574468085), broadcast(2.9850746268656714)) * Porosity times concentration [mol.m-3](Negative electrode porosity times concentration [mol.m-3], Separator porosity times concentration [mol.m-3], Positive electrode porosity times concentration [mol.m-3]), 0.01))) * grad(concatenation(broadcast(4.0), broadcast(2.127659574468085), broadcast(2.9850746268656714)) * Porosity times concentration [mol.m-3](Negative electrode porosity times concentration [mol.m-3], Separator porosity times concentration [mol.m-3], Positive electrode porosity times concentration [mol.m-3]))) - grad(Electrolyte potential [V](Negative electrolyte potential [V], Separator electrolyte potential [V], Positive electrolyte potential [V]))'], domains={'primary': ['negative electrode', 'separator', 'positive electrode'], 'secondary': ['current collector']})

During handling of the above exception, another exception occurred:

KeyError                                  Traceback (most recent call last)
[... 162 lines cut ...]
    731     self._discretised_symbols[symbol] = discretised_symbol
    732     discretised_symbol.test_shape()

File ...\pybamm\discretisations\discretisation.py:765, in Discretisation._process_symbol(self, symbol)
    762 if isinstance(symbol, pybamm.BinaryOperator):
    763     # Pre-process children
    764     left, right = symbol.children
--> 765     disc_left = self.process_symbol(left)
    766     disc_right = self.process_symbol(right)
    767     if symbol.domain == []:

File ...\pybamm\discretisations\discretisation.py:732, in Discretisation.process_symbol(self, symbol)
    730 discretised_symbol = self._process_symbol(symbol)
    731 self._discretised_symbols[symbol] = discretised_symbol
--> 732 discretised_symbol.test_shape()
    734 # Assign mesh as an attribute to the processed variable
    735 if symbol.domain != []:

File ...\pybamm\expression_tree\symbol.py:977, in Symbol.test_shape(self)
    975     self.shape_for_testing
    976 except ValueError as e:
--> 977     raise pybamm.ShapeError("Cannot find shape (original error: {})".format(e))

ShapeError: Cannot find shape (original error: operands could not be broadcast together with shapes (41,1) (60,1) )
```

ejfdickinson, 2023-12-18:
@rtimms Feels like this is a side effect of #3257 where an unidentified breaking change has crept in. Is it the case that in this parameter definition structure, ambient temperature now _has_ to be f(y,z,t), and can't just be f(t)?

rtimms, 2023-12-18:
Correct, you need to wrap the `Interpolant` in a function of (y,z,t)

rtimms, 2023-12-18:
This should have been documented as a breaking change. I don't know if there is some nice way to automatically detect if it is just a function of a single variable, in which case it is assumed to be a function of time only.

ejfdickinson, 2023-12-18:
@rtimms If the thermal model is "isothermal" or "lumped", isn't every f(y,z,[vars]) by definition f([vars])?

It feels like this drives required usage in the opposite direction from the geometry-agnosticism defined for these thermal models, also in #3257.

rtimms, 2023-12-18:
Yes, that is a fair point. At the moment, every `FunctionParameter` is defined as a function of the variables it _could_ be a function of, in any model/geometry, so we just define the function once in the parameters class. This is ok for variables that depend on other variables, e.g. D(c,T), but I agree shouldn't be the case for variables that depend on space and time. We should define these differently based on the model geometry so that we can throw an explicit and informative error.

rtimms, 2023-12-18:
Historically, most of the thermal stuff in PyBaMM was developed for single-layer pouch cell models, so there are lots of hangovers from design choices based on that.

ejfdickinson, 2023-12-18:
I'd feel it's most Pythonic to put optional arguments last, and explicitly allow them to be optional. In general we don't want to have to write functions whose return values don't depend on their arguments, right?

What's happened here is that you've enabled complicated behaviour by requiring simple behaviour to have a complicated calling pattern!

rtimms, 2023-12-18:
```python3
#
# Example showing how to solve the DFN with a varying ambient temperature
#

import pybamm
import numpy as np

pybamm.set_logging_level("DEBUG")

# load model
options = {"thermal": "lumped"}
model = pybamm.lithium_ion.DFN(options)

# create geometry
geometry = model.default_geometry

# load parameter values and process model and geometry
times = np.arange(0, 1810, 10)
tmax = max(times)

def ambient_temperature(y, z, t):
    return pybamm.Interpolant(times, 298.15 + 20 * (times / tmax), pybamm.t)

param = model.default_parameter_values
param.update(
    {"Ambient temperature [K]": ambient_temperature}, check_already_exists=False
)
param.process_model(model)
param.process_geometry(geometry)

# set mesh
var_pts = {"x_n": 30, "x_s": 30, "x_p": 30, "r_n": 10, "r_p": 10}
mesh = pybamm.Mesh(geometry, model.default_submesh_types, var_pts)

# discretise model
disc = pybamm.Discretisation(mesh, model.default_spatial_methods)
disc.process_model(model)

# solve model
t_eval = np.linspace(0, 3600 / 2, 100)
solver = pybamm.CasadiSolver(mode="fast", atol=1e-6, rtol=1e-3)
solution = solver.solve(model, t_eval)

# plot
plot = pybamm.QuickPlot(
    solution, ["X-averaged cell temperature [K]", "Ambient temperature [K]"]
)
plot.dynamic_plot()
```

rtimms, 2023-12-18:
That works, but I get an error with your example. Will look into it.

ejfdickinson, 2024-01-15:
@rtimms Any further insight into why the fix is not effective for the test case I suggested?

rtimms, 2024-01-22:
@ejfdickinson I'll make it a priority to look this week