pybamm-team/PyBaMM #3631: Define `FunctionParameters` that depend on space differently depending on geometry option

rtimms, 2023-12-18:
Some `FunctionParameters` depend on space and time (e.g. [ambient temperature](https://github.com/pybamm-team/PyBaMM/blob/256747c82e3a355fe20b8f577eb0b69480cad720/pybamm/parameters/thermal_parameters.py#L43)) but it doesn't always make sense to allow these to depend on space, for example if the thermal model is lumped then the temperature cannot depend on space. For the user, this can be confusing (see https://github.com/pybamm-team/PyBaMM/issues/3630). 

Is there a nice way to be able to define these functions differently depending on the model options? Then for lumped models users can pass `T(t)` and if they try to pass `T(y,z,t)` we can catch it and throw an informative error.

rtimms, 2023-12-18:
From @ejfdickinson

> I'd feel it's most Pythonic to put optional arguments last, and explicitly allow them to be optional. In general we don't want to have to write functions whose return values don't depend on their arguments, right?

What's happened here is that you've enabled complicated behaviour by requiring simple behaviour to have a complicated calling pattern!

Yes, this should work. Let's do that

rtimms, 2023-12-18:
In the actual model definitions, we will need to update the calls from `(y,z,t)` to `(t, y=y, z=z)` (best to be explicit in the latter in anticipation of other coordinate systems).

cringeyburger, 2023-12-18:
Hi! I would like to take this on.

Rishab87, 2025-02-27:
I would like to work on this issue, can this be assigned to me?