pybamm-team/PyBaMM #3647: PEP 723 support for SuiteSparse and SUNDIALS installation script

agriyakhetarpal, 2023-12-20:
### Description

[PEP 723](https://peps.python.org/pep-0723/) is one of the latest accepted Python PEPs which brings a `pyproject.toml`-like format for Python scripts for easy sharing and collaboration, where dependencies, supported Python versions, and other metadata can be added in TOML-style comments at the top of the script.

I have been advised by the developers to wait until support for tooling around this PEP gets mature in the coming year by packages like `pip-tools`, `pipx`, `virtualenv`, and others; so this would be a low-priority change for now and is being opened for tracking purposes. In conversation with Martin, we agreed that a providing a URL-based short link from the [pybamm.org](https://pybamm.org/) website pointing to the raw content link of the script can be used by beginner contributors and new developers to set up the build-time dependencies for the IDAKLU solver, and the next step would be to clone the repository and simply type `pip install -e ."[all]"` for a successful PyBaMM installation.

### Motivation

Easier installation of SUNDIALS and SuiteSparse when building PyBaMM from source for both Linux and macOS users from `pipx`, which can run remote URLs – say, something like `pipx run https://pybamm.org/install-builddeps.py`.

### Possible Implementation

A comment block at the top of the `install_KLU_Sundials.py` file

```python
# /// pyproject
# [run]
# requires-python = "<=3.9"
# dependencies = [
#   "wget",
#   "cmake",
# ]
# ///
```
which can be used by script runners to detect script metadata (and possibly install the dependencies too someday) along with other associated changes to the script to make it work as expected

### Additional context

https://discuss.python.org/t/pep-722-723-decision/36763

arjxn-py, 2024-01-12:
@agriyakhetarpal We've to keep this issue open, right?

agriyakhetarpal, 2024-01-12:
Yes, reopening as discussed in #3716. We have the table that specifies metadata for this script through the changes in that PR, the next step to be looked at at a later time would be how well Python tools like `pipx` are customising behaviour for this PEP.

agriyakhetarpal, 2024-03-30:
This can be closed because we might not have a `pybamm-requires` session anymore after the migration to `scikit-build-core` (#3564). If that doesn't work out, we can always reopen it.

PR #3716 (2024-01-12): Add PEP 723 support for SuiteSparse and SUNDIALS installation script
# Description

Introduce a TOML-style comment block at the top of the `install_KLU_Sundials.py` script. This block contains essential metadata, including the required Python version and a list of dependencies. This addition aligns the script with the PEP 723 guidelines, enhancing readability and portability for script runners and developers.

The metadata includes:
- The Python version requirement (<=3.9)
- Dependencies required for the script (wget, cmake)
- Additional information like the repository and documentation links
 [... cut]
commit fdbd8865 (2024-01-12): Add PEP 723 support for SuiteSparse and SUNDIALS installation script

Introduce a TOML-style comment block at the top of the `install_KLU_Sundials.py` script. This block contains essential metadata, including the required Python version and a list of dependencies. This addition aligns the script with the PEP 723 guidelines, enhancing readability and portability for script runners and developers.

The metadata includes:
- The Python version requirement (<=3.9)
- Dependencies required for the script (wget, cmake)
- Additional information like the repository and documentation links

This enhancem [... cut]
closed 2024-03-30 by agriyakhetarpal