pybamm-team/PyBaMM #3649: [Bug]: Geometry parameters must be Scalars after parameter processing

Mrzhang-hub, 2023-12-21:
### PyBaMM Version

23.9

### Python Version

3.9

### Describe the bug

ValueError: Geometry parameters must be Scalars after parameter processing.

### Steps to Reproduce

import pybamm
model = pybamm.lithium_ion.SPMe()
param = pybamm.ParameterValues('Prada2013')
param['Current function [A]'] = 1*param['Nominal cell capacity [A.h]']
param.update({'Positive particle radius [m]': "[input]"})
sim = pybamm.Simulation(model=model,parameter_values=param)
sol = sim.solve([0,3600],inputs={'Positive particle radius [m]':5e-08},calculate_sensitivities=True)
J = sol['Voltage [V]'].sensitivities['Positive particle radius [m]']

tinosulzer, 2023-12-21:
Geometry parameters (particle radius and thicknesses) don't work with inputs at the moment unfortunately. This is a known issue but challenging to fix

Jiyuan-Wangrwth, 2024-01-11:
I am also facing the same bug, there is any other way to make these variables?

BradyPlanden, 2024-01-18:
As Valentin mentioned above, the geometric parameters require a rebuild of the model when changes are made. As a result, the gradient information is not available. As far as I know, this is the only way to change the geometric parameters:

```python
import pybamm
model = pybamm.lithium_ion.SPMe()
param = pybamm.ParameterValues('Prada2013')
param['Current function [A]'] = 1*param['Nominal cell capacity [A.h]']
param.update({'Positive particle radius [m]': 5e-8})
sim = pybamm.Simulation(model=model,parameter_values=param)
sol = sim.solve([0,3600])
```
In PyBOP we have a [PR open](https://github.com/pybop-team/PyBOP/pull/169) to allow fitting of these parameters, which essentially rebuilds the required model objects at each iteration. You may be able to find some useful information related to this challenge on that branch.

Jiyuan-Wangrwth, 2024-01-19:
This way is great, which assign the parametervalue directly with avoiding "inputs". 
But  how to apply it in liionpack?
e.g.
`import liionpack
import pybamm
import numpy as np`
`Np,Ns=1,96
net= liionpack.setup_circuit(Np=Np, Ns=Ns) `
`parametervalues=pybamm.ParameterValues("Chen2020")`
`variable=5.56e-05*np.ones(96)`
`variable[:50]=6.56e-05`
`parametervalues.update(
    {
        "Negative electrode thickness [m]": variable,
    }
)`
`experiment= pybamm.Experiment([("Discharge at 5 A for 700 s", 
                                "Charge at 5 A for 100 s")]*2, period="10 s")`
`output= liionpack.solve(
                       netlist=net,
                       parameter_values=parametervalues,
                       experiment=experiment,
                       sim_func=liionpack.basic_simulation,
)`
`liionpack.plot_output(output)`
there will be an error `TypeError: Cannot process parameter`
but it works.
`import liionpack`
`import pybamm`
`import numpy as np`
`Np,Ns=1,96
net= liionpack.setup_circuit(Np=Np, Ns=Ns) `
`parametervalues=pybamm.ParameterValues("Chen2020")`
`variable=5.56e-05*np.ones(96)`
`parametervalues.update(
    {
        "Negative electrode thickness [m]": 5.56e-05,
    }
)`
`experiment= pybamm.Experiment([("Discharge at 5 A for 700 s", 
                                "Charge at 5 A for 100 s")]*2, period="10 s")`
`output= liionpack.solve(
                       netlist=net,
                       parameter_values=parametervalues,
                       experiment=experiment,
                       sim_func=liionpack.basic_simulation,
)`
`liionpack.plot_output(output)`
maybe because variable is an array rather than a spezifical number.  
So when I want to set these cells in different paramter values(radius or thickness), how to make it possible?

aabills, 2024-12-11:
Reopening because we are fixing this now.

PR #4665 (2025-01-02): Geometric input params
# Description

This pull request implements a `SymbolicUniform1DSubmesh`, which allows for geometric parameters, such as electrode thicknesses and particle sizes to be `pybamm.InputParameter`'s. 

Fixes #3649 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end ch [... cut]
closed 2025-01-02 by aabills