pybamm-team/PyBaMM #3660: [Bug]: example notebooks are failing

agriyakhetarpal, 2023-12-24:
### PyBaMM Version

`develop`

### Python Version

3.11

### Describe the bug

Due to an internal `nbmake` error – I think this was introduced by the changes #3492 since it failed twice there but was passing earlier. I don't think code formatting changes can cause such an error but I may be mistaken.

### Steps to Reproduce

Run example tests on recent PRs

agriyakhetarpal, 2023-12-24:
I tested locally and they fail there as well, @Saransh-cpp @arjxn-py could you check too?

Saransh-cpp, 2023-12-24:
I can reproduce this locally, but I get the exact same logs if I revert the `ruff format` commits. The logs are very hard to debug as almost all of them are originating from `nbmake` and not `pybamm`.

I think this is an `nbmake` issue, might get fixed after the holidays.

agriyakhetarpal, 2023-12-24:
This is strange, they pass for me locally if I revert the ruff-format changes

Saransh-cpp, 2023-12-24:
That's weird. I cleaned up my nox environments and did a fresh run, still failed with the same error. Maybe @arjxn-py can help with reproducing this.

arjxn-py, 2023-12-24:
Yes, trying to reproduce it locally while `ruff` commits reverted right now. Getting back in sometime.

agriyakhetarpal, 2023-12-25:
I think the problem is not with any individual notebook since running any of the failing notebooks separately does not cause an issue. The notebooks that are failing are different every time and I am able to reproduce it both before and after the `ruff format` changes now. 

The error trace is a bit difficult to debug, but it doesn't look like an error but rather a warning about garbage collection – maybe disabling these warnings through `pytest` can help

PR #3665 (2023-12-27): Ignore internal nbmake warnings - fix notebook tests
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #3660

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Op [... cut]
closed 2023-12-27 by Saransh-cpp