pybamm-team/PyBaMM #3664: Utilise caches for Windows nightly tests and  test wheels weekly instead of fortnightly

agriyakhetarpal, 2023-12-25:
The scheduled tests (unit, integration, examples) are run in a CRON job overnight for all platforms, but they don't compile the IDAKLU solver for Windows. We don't do it in PR tests to save time, but it should be reasonable to do so for the nightly tests – I have noticed that it takes around 15 minutes to compile the build-time dependencies on Windows, and less than three minutes if a `vcpkg` cache hit occurs. Allowing this cache to be set for the nightly tests on an ephemeral system like this would make sure that it will remain available for further tests and when one tries to debug the wheels or publish them, and thereby ensure that the jobs remain speedy. The cache can be invalidated by any changes to the `vcpkg` configuration files anyway so it is ensured that stale caches will never be used.

2. Building wheels weekly instead of fortnightly ensures that the set cache can be retained for further use in different scenarios, since GitHub Actions caches are evicted after a duration of one week.

R-Yash, 2024-01-31:
@agriyakhetarpal I want to work on this issue. I am thinking of modifying the `run_periodic_tests.yaml` by adding a step to cache vkpg dependencies and changing the tests to weekly in `publish_pypi.yml`. Is this the correct approach?

agriyakhetarpal, 2024-01-31:
Yes, that's right. I shall assign you this issue; please feel free to seek any more information as you need it.

R-Yash, 2024-01-31:
Thanks @agriyakhetarpal

agriyakhetarpal, 2024-04-26:
Closing because it's most likely not needed at this time (testing every two weeks is fine, caches are not needed for nightly Windows tests right now); can be re-opened sometime later.

closed 2024-04-26 by agriyakhetarpal