pybamm-team/PyBaMM #3666: Reduce the number of Docker Images

arjxn-py, 2023-12-26:
Currently we build and push 5 variants of Docker Images regularly i.e. `latest`, `jax`, `odes`, `idaklu` & `all`.
The idea behind this issue is to have a single standard image variant to be updated regularly to save resources and make things simple for developers.

agriyakhetarpal, 2023-12-26:
Removed the `needs-reply` label lest the issue gets closed automatically after thirty days. I renamed the title instead to explicitly mark it as a discussion

agriyakhetarpal, 2024-01-18:
Prior to starting with #3692, I think we should arrive at a consensus here so as to reduce coordinated effort.

<br>

Reiterating from #3661 for more visibility, the current reasoning for reducing the number of Docker images is as follows:

1. The `pybamm/pybamm:latest` image, i.e., without any solvers is just the same as cloning the GitHub repository or running `pip install git+https://github.com/pybamm-team/pybamm`, since it does not require any compiled steps. The procedure of installing PyBaMM without `pybind11` present for linkage is essentially going to skip the IDAKLU extension, without which PyBaMM is a pure-Python package.
2. The `pybamm/pybamm:jax` image just installs the JAX-based solvers, whose installation is very fast owing to the presence of wheels on PyPI. There is just one command extra for this purpose, which is just `pip install pybamm[jax]` – and with #3658, it might make sense to provide the IDAKLU solver and the JAX solver together instead of in separate images.
3. `pybamm/pybamm:odes` (i.e., the `ODES` argument) re-uses the SUNDIALS installation script that we use for the compilation of the IDAKLU solver and just points `SUNDIALS_INST` to the required path.
4. In a nutshell, the IDAKLU image is the most important one to target and maintain, and IMO the only one we should maintain. We can explore reducing the build-time for the Docker image through the use of aggressive build caching (possible for tools like `poetry`, will have to explore other methods) and make the ALL image the default one, which would then contain the IDAKLU, `scikits.odes` (re-uses the SUNDIALS compiled for the IDAKLU solver), and the JAX solvers.

This should simplify the tags for the image too – we just have to support `pybamm/pybamm:latest` (and `pybamm/pybamm:vYY.M` after #3661 gets completed).

agriyakhetarpal, 2024-02-23:
It was decided in the last developer meeting on 19th February that we can indeed reduce the number of Docker images, and establish just one image to contain all three solvers (`scikits.odes`, `JaxSolver`, and `IDAKLU`) and remove the image without the solvers. This can be taken up with #3692, I can revisit it this week and then get #3661 unblocked.

agriyakhetarpal, 2024-02-25:
Assigning @santacodes to this PR as discussed in #3692

agriyakhetarpal, 2024-02-25:
I can't assign @santacodes to this issue apparently, but this is being worked on along with #3692

santacodes, 2024-04-10:
Working on a PR, with some changes in dockerfile, docker workflow and docs. Hopefully will open it by tomorrow.

PR #3992 (2024-04-14): Reduced the number of docker images by one `pybamm:latest`
# Description

The `Dockerfile` is reduced to include all the solvers by default, the documentation for `install from source (docker)` and `docker.yml` workflow file have been updated to push only one single docker image tagged `pybamm:latest` to dockerhub with all the solvers contained in them by default, i.e `IDAKLU` and `JAX`.

Fixes #3666 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note rever [... cut]
PR #3901 (2024-08-07): Migrated from miniconda to official python bullseye docker image
# Description

This draft is in continuation to #3874 which was closed due to messy branches and the image failing to build IDAKLU solver on aarch64 linux. Initially the idea was to migrate to a single docker image of manylinux2014 which was compliant to [PEP599](https://peps.python.org/pep-0599/), but due to some complications, the official python image along with venv were decided upon implementation, check #3874 discussions for more information.

Fixes #3692
Fixes #3666 

## Type of change

Please add a line in the relevant se [... cut]
closed 2024-04-14 by arjxn-py