pybamm-team/PyBaMM #3668: [Bug]: There is an error in the 'Getting Started' section when we click the link in the example notebooks

Abhishekjsr283204, 2023-12-27:
### PyBaMM Version

PyBaMM version

### Python Version

python version

### Describe the bug

[image]
In this  above section
[image]
This error we got when we clicked this

### Steps to Reproduce

[image]
In this  above section
[image]
This error we got when we clicked this

agriyakhetarpal, 2023-12-27:
Hi @Abhishekjsr283204, the example notebooks have been moved to the [`docs/source/examples/` folder](https://github.com/pybamm-team/PyBaMM/tree/develop/docs/source/examples/notebooks) and the [PyBaMM website](https://pybamm.org) will be used to host the contents of the PyBaMM Wiki going forward. In specific, the page you are looking at is https://pybamm.org/gsoc/2023/, which has an updated link to the requested source. I hope this helps!

closed 2023-12-27 by agriyakhetarpal

Abhishekjsr283204, 2023-12-27:
okay got it thank you for response