pybamm-team/PyBaMM #3669: ERROR:pybamm.logger:Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:

Akila1993, 2023-12-28:
Hi 

When I run the [Modelling coupled degradation mechanisms in PyBaMM](https://docs.pybamm.org/en/latest/source/examples/notebooks/models/coupled-degradation.html) for 500 cycles it gives me the following error. 

At t = 292.526 and h = 4.54225e-10, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 294.731 and h = 4.53026e-10, the corrector convergence failed repeatedly or with |h| = hmin.
2023-12-28 12:11:10.017 - [ERROR] callbacks.on_experiment_error(224): Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_CONV_FAIL". Consult IDAS documentation.
ERROR:pybamm.logger:Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_CONV_FAIL". Consult IDAS documentation.

This is the same in google colab and the locally in my computer. 

Thank!

DrSOKane, 2024-01-03:
Happy new year Akila!

I tested this code for 1000 cycles but not for 500, so try again for 1000 cycles. If it still doesn't work, I'll look into it further.

Akila1993, 2024-01-03:
Hi Dr @DrSOKane , 

Thank you for your respond and wishing you also a happy new year. 

Yes, I have tried it for 1000 cycles and it gives me the same error as below. I hope you will have a look about it. 

Thank you for your consideration. 

At t = 287.442 and h = 1.18304e-10, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 292.529 and h = 5.72136e-10, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 292.45 and h = 5.75579e-10, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 297.728 and h = 4.76176e-10, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 299.997 and h = 1.51245e-10, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 277.016 and h = 3.56316e-10, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 287.293 and h = 2.05602e-10, the corrector convergence failed repeatedly or with |h| = hmin.
2024-01-03 22:49:55.923 - [ERROR] callbacks.on_experiment_error(224): Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_CONV_FAIL". Consult IDAS documentation.
ERROR:pybamm.logger:Simulation error: Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_CONV_FAIL". Consult IDAS documentation.

ldavezac, 2024-01-04:
Hi, 
Happy new year. 
I faced the issue when using the DFN model  for simulation: 
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDASolve returned "IDA_CONV_FAIL". Consult IDAS documentation.
I will be interested if any informations about it are found. Thank you!

DrSOKane, 2024-01-04:
After several hours of trial and error, I managed to track down the source of the problem. It's to do with the experiment protocol.

The original experiment was this:

```
exp = pybamm.Experiment(
    [
        "Hold at 4.2 V until C/100 (5 minute period)",
        "Rest for 4 hours (5 minute period)",
        "Discharge at 0.1C until 2.5 V (5 minute period)",  # initial capacity check
        "Charge at 0.3C until 4.2 V (5 minute period)",
        "Hold at 4.2 V until C/100 (5 minute period)",
    ]
    + [
        (
            "Discharge at 1C until 2.5 V",  # ageing cycles
            "Charge at 0.3C until 4.2 V (5 minute period)",
            "Hold at 4.2 V until C/100 (5 minute period)",
        )
    ]
    * cycle_number
    + ["Discharge at 0.1C until 2.5 V (5 minute period)"],  # final capacity check
)
```

_i.e._ everything except for the 1C discharge had a 5 minute period, whereas the 1C discharge itself had the default 1 minue period.

The code was subsequently changed so that everything, including the 1C discharge, had a 5 minute period. When I tried restoring the original experiment protocol, it worked. I am shocked that such a small change would make such a difference, but after years of experience with PyBaMM I am no longer surprised.

I will make a pull request right away to revert the change to the notebook so that future users won't have this problem.

Simon

PR #3691 (2024-01-05): Degradation example update
The experiment protocol for the coupled degradation example notebook was changed so that the 5 minute period applied to all stepts of all cycles, including the 1C discharge. This results in the simulation being unstable for large numbers of cycles. This pull request reverts the change and fixes the issue.

Fixes #3669 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of  [... cut]
closed 2024-01-05 by kratman