pybamm-team/PyBaMM #3679: `ParameterValues.create_from_bpx` - support BPX v0.4

ejfdickinson, 2024-01-03:
### Description

Add support to `ParameterValues.create_from_bpx()` for new [BPX 0.4](https://bpxstandard.com/bpx-standard/) features.

- Import blended electrode parameters according to specification.
- Consider whether PyBaMM could define a standard for importing the necessary branches for `current sigmoid` hysteresis from user-defined BPX `FloatFunctionTable` features.

### Possible Implementation

A simple PyBaMM-internal standard might be that the contents of the BPX user-defined parameters node is read directly into the `ParameterValues` object. Then users could add any PyBaMM-compatible parameter names to the BPX, under this node.

Since the content of a BPX file implies the type of model that should be used (DFN vs SPM, blended electrode or not, hysteresis or not), could `create_from_BPX` optionally return a `dict` of appropriate model options? Or else another utility function could do this?

rtimms, 2024-01-10:
See https://github.com/pybamm-team/PyBaMM/pull/3414

rtimms, 2024-01-10:
This was working when I made the draft PR. I need to check it works with the actual v.0.4.0 release. Will try to get to it this week so it can be included in the final 24.1 release of PyBaMM (tagging @Saransh-cpp)

rtimms, 2024-01-10:
> A simple PyBaMM-internal standard might be that the contents of the BPX user-defined parameters node is read directly into the ParameterValues object. 

This is the implementation in the draft PR

> Since the content of a BPX file implies the type of model that should be used (DFN vs SPM, blended electrode or not, hysteresis or not), could create_from_BPX optionally return a dict of appropriate model options? Or else another utility function could do this?

Yes, but this might require us to be strict about how/where people give this in the BPX file. E.g. if hysteresis is given as a user-defined parameter, should we expect users to explicitly say this in the description somewhere? I wouldn't like to try and infer the model option from user-defined parameters.

ejfdickinson, 2024-01-11:
Thanks @rtimms - I think you are right to discard the second suggestion, and leave this to the user. Probably it should be documented that `ParameterValues.create_from_bpx()` only creates the `pybamm.ParameterValues` object, and does not embed aspects of the BPX standard that oblige a certain _model_, such as DFN vs SPM.

I advise that this issue can be closed if the draft PR #3414 is tested and merged.

PR #3414 (2024-01-25): [WIP] updates for BPX v0.4.0 compatibility 
# Description
Add support for blended electrodes and user-defined parameters in BPX. Create parameters from BPX file in a loop using `partial`.

PyBaMM now requires `bpx>=0.4.0`

Note: BPX set initial concentrations based on SOC, which is not yet supported in pybamm for blended electrodes (see  https://github.com/pybamm-team/PyBaMM/issues/2682). If a user supplies a "blended electrode" BPX file then they must manually provide initial concentrations after the `ParameterValues` class has been created (or supply them in the "user-defined" section).

Fix [... cut]
closed 2024-01-25 by rtimms