pybamm-team/PyBaMM #3692: Migrate from `conda` to `venv` inside Docker Images

arjxn-py, 2024-01-06:
This would potentially cut down an external system dependency `conda` & would reduce the inconsistency of environments across different OS for IDAKLU solver images.

arjxn-py, 2024-01-06:
Marking it easy at the moment but since the base image would be changed as well, the complexity might reach `medium` as well.

agriyakhetarpal, 2024-01-06:
It might reach `medium`, the newer choice of base image should be GLIBC compliant. If it's possible, we can use the checked out files from the repository to build the image instead of cloning PyBaMM inside it.

agriyakhetarpal, 2024-01-16:
I was thinking of taking this up, if you don't mind that @arjxn-py – been wanting to learn more things about Docker

agriyakhetarpal, 2024-01-16:
I was thinking of taking this up if you don't mind that, @arjxn-py – I have been been wanting to learn more things about Docker recently

santacodes, 2024-02-25:
Can I take this issue if it's not being worked on by any chance?

agriyakhetarpal, 2024-02-25:
Sure, @santacodes – I won't be able to work on this realistically anytime soon, feel free to take this up. Some guidelines are:

1. We need this image for both amd64 and aarch64 platforms.
2. The image needs to be compliant with the PEP599 `manylinux2014` platform tag, i.e., the same one as what we publish PyPI wheels for – please view the document here: https://peps.python.org/pep-0599/. This means that it should rely on a GLIBC version no later than `2.17`.
3. We should target #3666 to reduce the number of images and include one image with the `latest` with all solvers (the current `latest` image includes no compiled solvers). Feel free to close both issues in one PR in order to reduce merge conflicts.

PR #3901 (2024-08-07): Migrated from miniconda to official python bullseye docker image
# Description

This draft is in continuation to #3874 which was closed due to messy branches and the image failing to build IDAKLU solver on aarch64 linux. Initially the idea was to migrate to a single docker image of manylinux2014 which was compliant to [PEP599](https://peps.python.org/pep-0599/), but due to some complications, the official python image along with venv were decided upon implementation, check #3874 discussions for more information.

Fixes #3692
Fixes #3666 

## Type of change

Please add a line in the relevant se [... cut]
closed 2024-08-07 by arjxn-py