pybamm-team/PyBaMM #3720: [Bug]: Temperature interpolant does not work for v23.9

WoohyunJeong95, 2024-01-12:
### PyBaMM Version

23.9

### Python Version

3.11

### Describe the bug

I tried to give temperature measurement as an input to SPMe model. But interpolant gives me a shape error. It worked fine with SPM (v23.9) and also SPMe with v23.5 worked fine. Is there any way to use interpolant for temperature for v23.9? 

### Steps to Reproduce

import pybamm
import numpy as np

t_history = np.arange(0,3600)
T_history = np.ones(3600)*300

param = pybamm.ParameterValues("Chen2020")
amb_T_K = pybamm.Interpolant(t_history,T_history, pybamm.t)
param.update({'Ambient temperature [K]': amb_T_K})
model = pybamm.lithium_ion.SPMe()
sim = pybamm.Simulation(model, parameter_values=param)
solution = sim.solve([0,3600])

kratman, 2024-01-15:
@WoohyunJeong95 Can you update your bug report with the error messages from your simulation?

WoohyunJeong95, 2024-01-16:
Hi, Thank you for your answer. Here is the error messages I got with version 23.9. 

[image]
[image]
[image]
[image]
[image]
[image]

rtimms, 2024-01-16:
Duplicate of https://github.com/pybamm-team/PyBaMM/issues/3630

closed 2024-01-16 by rtimms

WoohyunJeong95, 2024-01-16:
I figured out that I can make this work by using options={"thermal": "lumped"}. 
The reason I want to use temperature interpolation for ambient temperature is to set the cell temperature, as I already have cell temperature measurements. So this can be done by using very high heat transfer coefficient value so that cell temperature directly becomes ambient temperature. 
I attach the code for those who are interested. 

import pybamm
import numpy as np

t_history = np.arange(0, 3600, 10)
tmax = max(t_history)
T_history = 298.15 + 20 * (t_history / tmax)

param = pybamm.ParameterValues("Chen2020")

amb_T_K = pybamm.Interpolant(t_history,T_history, pybamm.t)

param.update({'Initial temperature [K]':T_history[0]})
param.update({'Ambient temperature [K]': amb_T_K})
param.update({'Total heat transfer coefficient [W.m-2.K-1]':10000}) # Very high value to make the cell temperature same as ambient temperature

model = pybamm.lithium_ion.SPMe(options={"thermal": "lumped"})
sim = pybamm.Simulation(model, parameter_values=param)
solution = sim.solve([0,3600])

sim.plot(output_variables=['Ambient temperature [K]','X-averaged cell temperature [K]'])

[image]

rtimms, 2024-01-26:
This is now fixed, see https://github.com/pybamm-team/PyBaMM/pull/3761