pybamm-team/PyBaMM #3722: API improvements to `plot_voltage_components`

brosaplanella, 2024-01-14:
### Description

When reviewing the getting started notebooks I have spotted a few small things we could implement to improve the API of `plot_voltage_components`.  Ideas:

- Allow `plot_voltage_components` to take a `Simulation` in addition to a `Solution` (current behaviour)
- Allow the syntax `simulation.plot_voltage_components` and `solution.plot_voltage_components`

### Motivation

They do not add functionality but would make the method similar to `plot` (in terms of API) and would make the learning curve in the tutorials smoother.

R-Yash, 2024-01-15:
@brosaplanella I want to work on this issue. How do I get started?

brosaplanella, 2024-01-15:
For the first point, you want to add an if statement in [plot_voltage_components.py](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/plotting/plot_voltage_components.py) that, if the argument `solution` is a simulation, it takes the solution.

For the second point, you would need to add a method to `Solution` similar to `plot`:
https://github.com/pybamm-team/PyBaMM/blob/c3def31a04df163769851733657d9b49b3700bca/pybamm/solvers/solution.py#L545-L560

AlessioBugetti, 2024-01-15:
@R-Yash , I just realized that I missed your comment about wanting to work on this issue. My apologies for not seeing it earlier! I've already started working on it, not knowing about your interest.

I really appreciate your initiative and would love to collaborate or hear any ideas you might have. Again, sorry for the oversight, and looking forward to your input.

R-Yash, 2024-01-15:
That's fine @AlessioBugetti. I am a new contributor and still figuring things out in this repo. You can continue with your work on the issue.

closed 2024-01-18 by Saransh-cpp