pybamm-team/PyBaMM #3725: Check if the solution exists before creating a GIF (`Simulation.create_gif` and `BatchStudy.create_gif`)

Saransh-cpp, 2024-01-15:
I liked how #3723 checks if the simulations have been solved before proceeding ahead to the plotting functionality. It would be nice to have similar checks in -

https://github.com/pybamm-team/PyBaMM/blob/c3def31a04df163769851733657d9b49b3700bca/pybamm/batch_study.py#L184-L187

and 

https://github.com/pybamm-team/PyBaMM/blob/c3def31a04df163769851733657d9b49b3700bca/pybamm/simulation.py#L1047-L1050

prady0t, 2024-01-24:
Would like to take this on.

R-Yash, 2024-01-31:
Hi @prady0t are you still working on this issue?

prady0t, 2024-01-31:
yes

prady0t, 2024-01-31:
@Saransh-cpp BatchStudy class doesn't have an attribute `solution` unlike Simulation. How to check for empty solutions there?

Saransh-cpp, 2024-01-31:
You can check if the `self.sims` list is empty and throw an error if it is -

https://github.com/pybamm-team/PyBaMM/blob/c3def31a04df163769851733657d9b49b3700bca/pybamm/batch_study.py#L116

The simulations are being initialised and solved in the `solve()` method, so this would be a good indication of if the user has called the `solve()` method before calling the `create_gif()` method.

prady0t, 2024-01-31:
why is 
https://github.com/pybamm-team/PyBaMM/blob/063a3830afd5547f7a78e283ae81058465cbdd3e/pybamm/batch_study.py#L116
declared outside `__init__` should'nt it be inside it?

PR #3795 (2024-02-02): Added test to check if simulation has been solved in `simulation.py` and `batch_study.py`
# Description

Simple check for if simulations have been solved or not.

Fixes #3725 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking [... cut]
closed 2024-02-02 by agriyakhetarpal