pybamm-team/PyBaMM #3743: Conserve `needs-reply` GitHub Actions workflow runs

agriyakhetarpal, 2024-01-18:
The `needs-reply` workflow currently runs on all issue comments or PR comments and it runs too much: https://github.com/pybamm-team/PyBaMM/actions/workflows/need_reply_remove.yml

I am not sure as to how to fix this as of now, and we don't mind it at all (GitHub Actions is completely free for public repositories) but it would be great if the runs for this workflow could be limited in some way or form. Some suggestions could be to:

1. Run only on stale conversations or on different events
2. Switch to a different GitHub App or GitHub Action
3. Skip the workflow on PRs
4. Remove the workflow altogether
5. and so on

santacodes, 2024-03-10:
Maybe we could schedule it to run on a specific interval using cron job but I am not sure how much of an interval would be ideal for this particular action.

agriyakhetarpal, 2024-03-10:
Yes, that makes sense. How much would you expect it should run?

I would say we can run it weekly on Sunday (or thrice weekly on Monday, Wednesday, and Friday perhaps or would that be too much?), and skip it on un-labelled issues. Also, it should not run at all on PRs, since we use the `needs-reply` label only for issues.

The reason for the addition of the `needs-reply label` was was to mark issues for closing after thirty days of labelling that are without a response after the duration, so I think the minimum cadence should be at least fortnightly.

santacodes, 2024-03-10:
A weekly run would be fair if the frequency of issues created are less, or we could maybe do it twice a week.

agriyakhetarpal, 2024-03-10:
A weekly run sounds good to me. We can also customise the granularity based on other GitHub Issue events (while keeping it as minimal as possible).

PR #3891 (2024-03-13): Added a schedule on `needs-reply remove` workflow 
# Description

Added a schedule on the [need_reply_remove](https://github.com/pybamm-team/PyBaMM/actions/workflows/need_reply_remove.yml) workflow on every Monday 03:00 hours to limit the resources that get utilized while running this action.

Fixes #3743 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (no [... cut]
closed 2024-03-13 by agriyakhetarpal