pybamm-team/PyBaMM #3763: New pybamm.Event type for changing a flag or variable

DrSOKane, 2024-01-24:
### Description

PyBaMM currently has four types of events that carry out specific functions. However, I'm working on a new project that requires a fifth type of event, which changes a specific flag or variable when hit.

### Motivation

I'm working with @smitasahu2 and @YoanaGrudeva on a model of hysteresis in lithium iron phosphate (LFP) electrodes, which will utilize the particle size distribution functionality. Their model runs in MATLAB and we are now keen to implement it in PyBaMM. Much of the model can be built using PyBaMM as it currently exists, but CasADi's lack of events support is a major obstacle to full implementation. The new event can also be used to improve the current sigmoid OCP model so that for zero current, PyBaMM will "remember" what the most recent current direction was. (Plett calls this the zero-order hysteresis model.)

rtimms, 2024-01-25:
@DrSOKane let me know if you'd like any support with this - better hysteresis models should be a high priority on the PyBaMM roadmap IMO. 

Also, not sure of the details of the implementation, but an alternative/more general approach would be Plett's SOC varying model which would then give you a decay time. The instant switch would be a limiting case. 

It could be implemented something like 
```python3
import pybamm
import matplotlib.pyplot as plt

model = pybamm.BaseModel()

t = pybamm.t
I = 2 * (t > 0) * (t < 1) - 2 * (t > 2) * (t < 3)

s = pybamm.Variable("s")
a = pybamm.InputParameter("a")
abs_aI = abs(a * I)
model.rhs[s] = -abs_aI * s - pybamm.sign(I) * abs_aI
model.initial_conditions[s] = 0

model.variables = {"t": t, "I": I, "s": s}

sim = pybamm.Simulation(model)

fig, ax = plt.subplots(1, 2)
for a_value in [5, 10, 50]:
    sim.solve([0, 4], inputs={"a": a_value})
    sol = sim.solution
    ax[0].plot(sol["t"].data, sol["I"].data)
    ax[1].plot(sol["t"].data, sol["s"].data, label="a = {}".format(a_value))
ax[1].legend()
```

which gives

[image]

DrSOKane, 2024-01-25:
@rtimms I saw the PR where that's being implemented. It's superior to the zero-order hysteresis model. However, the application Yoana and Smita have in mind requires an event state _s_, which is either 0 or 1. I suggested using a tanh function (pybamm.Sigmoid) but the concentration at which _s_ changes depends on _s_, so without events you end up with a transcendental equation for _s_. Is there a way to use PyBaMM to solve such a transcendental equation and find _s_ that way?

rtimms, 2024-01-25:
Does it not just become another algebraic constraint? 

Can you share the equation (or a simplified example)?

YoanaGrudeva, 2024-01-26:
@rtimms 
We have a differential-algebraic system in which the variables are $c, \mu, j_{tr}$. The chemical potential in a particle is given by  $\mu = f(c,s)$, where $c$ is concentration and the switch $s = 0, 1$ is a state that indicates whether a particle is currently phase-separated. In general, we have $f(c,s) = s \hat{f}(c)$ and also $\frac{dc}{dt} = g(c,\mu)$. 

When evolving the system, at each step, we choose the function for the chemical potential based on the value of the switch $s$. When a critical value of $c$ is reached from a specified direction, $s$ changes value.  

When handled in Matlab, Events stop the solver when the critical concentration is reached and restart with the new value in a while loop. As I understand, events are not native to CasADi's solvers?

DrSOKane, 2024-01-29:
I had a think about this and I think I found the equation for _s_:

[image]
[image]

DrSOKane, 2024-01-29:
I think we can close this issue for now, because we haven't been able to establish a good case for implementing such a feature. But it can be reopened if necessary, and it was a useful discussion. @rtimms @YoanaGrudeva @smitasahu2

closed 2024-01-29 by DrSOKane

rtimms, 2024-01-31:
Yes, the implementation you suggest should work in principle. How are you going to implement it now? With events?