pybamm-team/PyBaMM #3766: Complex numbers in PyBaMM

brosaplanella, 2024-01-25:
### Description

At the moment we do not have a class to deal with complex numbers in PyBaMM. Python already offers the functionality so might be easy to bring it in, but not sure if CasADI is able to deal with complex numbers.

### Motivation

This would enable including EIS simulations (and data fitting) in the future.

brosaplanella, 2024-01-25:
@pybamm-team/maintainers what do you think?

martinjrobins, 2024-01-25:
sadly I don't think Casadi supports complex numbers:

```python
from casadi import *

x = MX.sym("x")
f = x**2 + 10
f = Function('f',[x],\
           [x**2 + 10])
print(f(2))
print(f(2 + 3j))
```

```
14
Traceback (most recent call last):
  File "/home/mrobins/git/PyBaMM/test.py", line 8, in <module>
    print(f(2 + 3j))
  File "/home/mrobins/git/PyBaMM/env/lib/python3.10/site-packages/casadi/casadi.py", line 23363, in __call__
    ret = self.call(args)
  File "/home/mrobins/git/PyBaMM/env/lib/python3.10/site-packages/casadi/casadi.py", line 20021, in call
    return _casadi.Function_call(self, *args)
NotImplementedError: Wrong number or type of arguments for overloaded function 'Function_call'.
  Possible prototypes are:
    call(self,dict:DM,bool,bool)
    call(self,[DM],bool,bool)
    call(self,[SX],bool,bool)
    call(self,dict:SX,bool,bool)
    call(self,dict:MX,bool,bool)
    call(self,[MX],bool,bool)
  You have: '(Function,(complex))'
```

martinjrobins, 2024-01-25:
Longer term I think it would be good to move to an MLIR-based backend rather than casadi, this will enable complex numbers and gpu support in conjunction with the more robust sundials solver (rather than the jax solver).

brosaplanella, 2024-01-29:
In the EIS case the CasADI issue would be easy to bypass, as the system to solve would be algebraic (it is in the frequency domain) so we could get away with other solvers.

RohitP2005, 2025-03-04:
@martinjrobins  I guess we can proceed with the issue now if possible.

RohitP2005, 2025-03-04:
I think we can move forward with the issue now @martinjrobins @brosaplanella

brosaplanella, 2025-03-19:
Tagging @rtimms, as when I opened this issue PyBaMM-EIS was not maintained but now it is.

brosaplanella, 2026-04-09:
#5433 Makes this issue not necessary