pybamm-team/PyBaMM #3768: "Could not find parameter Chen2020" - Maybe a bug, maybe my setup

Alistair-Douglas, 2024-01-25:
### PyBaMM Version

pybamm 0.2.4.post3

### Python Version

3.12

### Describe the bug

Following tutorial 4, no 3rd party parameter sets are available. 
I have tried:

pip uninstall pybamm
pip install pybamm -u

This did not change the outcome.

I have also tried searching the directory with:

try:
    for item in os.listdir(parameters_dir):
        print(item)
except FileNotFoundError as e:
    print("Error: ", e)

Providing:

lead-acid
lithium-ion

Is there something I need to do? I am using the Microsoft store python as I was having compatibility issues at one point. 
Thank you very much for your time guys I appreciate I am probably doing something stupid and this piece of software you have created is incredible.

### Steps to Reproduce

Every time I follow tutorial 4, even copy and pasting, I get the same error message, including for other 3rd party parameters.

Follow tutorial 4, copy the code,

"Could not find parameter Chen2020"

### Relevant log output

```shell
#Initial Error Message

FileNotFoundError                         Traceback (most recent call last)
Cell In[5], line 7
      4 import numpy as np
      5 import matplotlib.pyplot as plt
----> 7 parameter_values = pybamm.ParameterValues("Chen2020")

File ~\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\pybamm\parameters\parameter_values.py:76, in ParameterValues.__init__(self, values, chemistry)
     73 if values is not None:
     74     # If base_parameters is a filename, load from that filename
     75     if isinstance(values, str):
---> 76         file_path = self.find_parameter(values)
     77         path = os.path.split(file_path)[0]
     78         values = self.read_parameters_csv(file_path)

File ~\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\pybamm\parameters\parameter_values.py:755, in ParameterValues.find_parameter(path)
    753     if os.path.isfile(trial_path):
    754         return trial_path
--> 755 raise FileNotFoundError("Could not find parameter {}".format(path))

FileNotFoundError: Could not find parameter Chen2020
```

kratman, 2024-01-25:
@Alistair-Douglas Is there a reason you are using a version of PyBaMM from 2020? The current stable release is 23.9

agriyakhetarpal, 2024-01-26:
Hi @Alistair-Douglas, this error that you're receiving is because PyBaMM 23.9 supports just Python 3.8–3.11, and the upcoming PyBaMM 24.1 stable release will add support for Python 3.12. You may switch to a lower Python version for the time being or consider using the 24.1 pre-release(s) via the `$ pip install --pre pybamm` command.

closed 2024-01-26 by agriyakhetarpal

Alistair-Douglas, 2024-01-26:
> @Alistair-Douglas Is there a reason you are using a version of PyBaMM from 2020? The current stable release is 23.9

@agriyakhetarpal Honestly I am not sure, I followed the tutorials, tried both pip install and pip upgrade pybamm so thought I was on the most current version, evidently not. Thank you very much.

Alistair-Douglas, 2024-01-26:
> @Alistair-Douglas Is there a reason you are using a version of PyBaMM from 2020? The current stable release is 23.9

@kratman Honestly I am not sure, I followed the tutorials, tried both pip install and pip upgrade pybamm so thought I was on the most current version, evidently not. Thank you very much I can address this now thank you.

agriyakhetarpal, 2024-01-26:
The issue was that with this old version we did not have an upper bound in place for the Python version, leading `pip` to recognise this as a compatible one. My opinion on such things is to delete such old releases that are not relevant anymore, but that is more of a radical one compared to solutions like `pip` displaying a warning on the version being very old – for which again one doesn't really have a metric for determining what counts as "old"