pybamm-team/PyBaMM #3770: Rename parameters from "electrode" to "particle"

julian-evers, 2024-01-26:
As "positive electrode diffusivity" is currently [renamed](https://github.com/pybamm-team/PyBaMM/pull/3624) to "positive particle diffusivity" it might be a good time to rename other parameters similarly as non-breaking changes with deprecation warnings, though that might not be perfect for film-type electrodes. How do you think about below parameter names?

- Initial concentration in positive _particle_ [mol.m-3]
- Maximum concentration in positive _particle_ [mol.m-3]
- Positive electrode Bruggeman coefficient (_particle_)
- Positive _particle_ charge transfer coefficient
- Positive _particle_ exchange-current density [A.m-2]

brosaplanella, 2024-03-19:
Was this already fixed by https://github.com/pybamm-team/PyBaMM/pull/3624?

julian-evers, 2024-03-29:
Many thanks for catching up on the issue, it was not fixed by #3624
<https://github.com/pybamm-team/PyBaMM/pull/3624>. Would it still be nice
to have this implemented or is it irrelevant with the upcoming parameter
refactor <https://github.com/pybamm-team/PyBaMM/issues/3909> and can be
closed as not planned?

tinosulzer, 2024-03-30:
We're reviewing how parameters are defined as part of #3909 . Let's discuss there

closed 2024-03-30 by tinosulzer