pybamm-team/PyBaMM #3782: [Bug]: stop battery model options warning

Jadhav-Sourabh, 2024-01-30:
### PyBaMM Version

23.9

### Python Version

3.11

### Describe the bug

How to stop battery model options warning, when you use lumped battery sub model?

### Steps to Reproduce

How to stop battery model options warning, when you use lumped battery sub model?

Jadhav-Sourabh, 2024-01-30:
.

kratman, 2024-01-30:
@Jadhav-Sourabh Can you share the warnings and example code?

Jadhav-Sourabh, 2024-01-30:
> @Jadhav-Sourabh Can you share the warnings and example code?

Run basic discharge simulation with DFN, Prada parameters set, lumped thermal with Casadi solver. You will get optional warning about needed parameters because of lumped thermal sub model (add cell cooling surface area, cell volume, total heat transfer coefficient, asking to use BatteryModelOptions)

Saswatsusmoy, 2024-01-31:
Can we like pass the options in a dictionary while building the model?

Something like 

`options = {
    "cell cooling surface area": x,
    "cell volume": y,
    "total heat transfer coefficient": z
}
model = pybamm.modelname(param, options)`

rtimms, 2024-01-31:
This is just a warning as in the latest version we changed which parameters are used in the lumped model - the lumped model always uses the lumped parameters.

In version 24.1 the parameters GeometricParameters.A_cooling and GeometricParameters.V_cell are now computed from the electrode heights, widths and thicknesses if the "cell geometry" option is "pouch" and from the parameters "Cell cooling surface area [m2]" and "Cell volume [m3]", respectively, otherwise. The warning is removed in 24.1 as this is how the lumped model used to calculate these quantities (albeit the implementation was different). 

So this isn’t a bug, just a warning to make sure the user is aware of which parameters are being used.

rtimms, 2024-01-31:
See https://github.com/pybamm-team/PyBaMM/pull/3707

closed 2024-01-31 by rtimms