pybamm-team/PyBaMM #3794: Parameter value of Chen2020

Jiyuan-Wangrwth, 2024-01-31:
### Description

The default value of 'Electrode width [m]' is 1.58, which is not consistent with the value in Paper, it should be 0.065 m.
`parametervalues["Electrode width [m]"]`
`1.58`

agriyakhetarpal, 2024-02-01:
Hello @Jiyuan-Wangrwth – thanks, I checked the paper and the electrode width is indeed mentioned as 0.065m, with the electrode length (which is the electrode height in PyBaMM) as 1.58m; it should be the opposite.

https://github.com/pybamm-team/PyBaMM/blob/df29f68fe5f10350f411857345017ca2aaa94971/pybamm/input/parameters/lithium_ion/Chen2020.py#L257

I'm not completely sure whether this was intentional or this was a bug – perhaps these values were interchanged at the time of the parameter set being implemented so most probably it was indeed the latter, so I'll leave that to someone else to discuss. Would you be willing to submit a PR to fix these?

rtimms, 2024-02-01:
This is just a difference in terminology. The Chen paper uses the names "length" and "width" to refer to the unrolled electrode dimensions, whereas PyBaMM uses "width" and "height" (partly because lots of the early PyBaMM work focussed on pouch cells). 

For models like SPM, DFN, ... where you are only modelling variations in the through-electrode and particle dimensions, all you really care about is the electrode cross-sectional area (height times width).

closed 2024-02-01 by rtimms

agriyakhetarpal, 2024-02-01:
Thanks for the clarification, @rtimms!