pybamm-team/PyBaMM #3797: [Bug]: The residual function failed at the first call.

flag2fish, 2024-02-02:
### PyBaMM Version

23.9

### Python Version

3.9.13

### Describe the bug

2024-02-01 18:03:47.086 - [WARNING] simulation.solve(620): Ignoring t_eval as solution times are specified by the experiment
2024-02-01 18:03:47.087 - [NOTICE] callbacks.on_cycle_start(172): Cycle 1/1 (51.500 us elapsed) --------------------
2024-02-01 18:03:47.088 - [NOTICE] callbacks.on_step_start(180): Cycle 1/1, step 1/1: Rest for 30 min
The residual function failed at the first call. 
The residual function failed at the first call. 
The residual function failed at the first call. 
The residual function failed at the first call. 
The residual function failed at the first call. 
The residual function failed at the first call. 
2024-02-01 18:03:47.527 - [ERROR] callbacks.on_experiment_error(224): Simulation error: Maximum number of decreased steps occurred at t=0.0 (final SolverError: 'Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at .../casadi/core/function.cpp:330:
.../casadi/interfaces/sundials/idas_interface.cpp:596: IDACalcIC returned "IDA_FIRST_RES_FAIL". Consult IDAS documentation.'). For a full solution try reducing dt_max (currently, dt_max=600) and/or reducing the size of the time steps or period of the experiment. Set `return_solution_if_failed_early=True` to return the solution object up to the point where failure occured.

### Steps to Reproduce

import pybamm
import numpy as np
import matplotlib.pyplot as plt
model = pybamm.lithium_ion.DFN(
    {   
        "thermal": "x-full", 
        "surface form": "differential", 
        "SEI":"solvent-diffusion limited",
        "SEI porosity change": "true",
        "lithium plating": "partially reversible",
        "lithium plating porosity change": "true",
        "particle mechanics":("swelling and cracking", "swelling only"),
        "SEI on cracks": "true",
        "loss of active material": "stress-driven",
        "calculate discharge energy": "true",        
    }
)
var_pts = {"x_n": 5, "x_s": 5, "x_p": 5, "r_n": 30, "r_p": 30,}
def get_basic_parameters():
    parameter_values = pybamm.ParameterValues("OKane2022")
    parameters = parameter_values
    return parameters

def get_thermal_parameters():
    thermal_parameters = {
        "Negative current collector surface heat transfer coefficient [W.m-2.K-1]": 0,
        "Positive current collector surface heat transfer coefficient [W.m-2.K-1]": 0,
        "Negative tab width [m]": 0,
        "Positive tab width [m]": 0,
        "Negative tab heat transfer coefficient [W.m-2.K-1]": 0,
        "Positive tab heat transfer coefficient [W.m-2.K-1]": 0,
        "Edge heat transfer coefficient [W.m-2.K-1]": 0,
    }
    return thermal_parameters

def update_parameters():
    basic_parameters = get_basic_parameters()    
    thermal_parameters = get_thermal_parameters()
    basic_parameters.update(thermal_parameters, check_already_exists = False)
    return basic_parameters

parameter_values = update_parameters()

def negative_ocp(sto):
    u_eq = (        
        0.6379
        + 0.5416 * np.exp(-305.5309 * sto)
        + 0.044 * np.tanh(-(sto - 0.1958) / 0.1088)
        - 0.1978 * np.tanh((sto - 1.0571) / 0.0854)
        - 0.6875 * np.tanh((sto + 0.0117) / 0.0529)
        - 0.0175 * np.tanh((sto-0.5692) / 0.0875)
    )
    return u_eq

def negative_OCP_entropic_change(sto, c_s_max):
    dEdT = (
        2.8957
        - 5.2972 * pybamm.cos(pybamm.arctan(1/(sto**2) - 1))
        + 4.0370 * pybamm.cos(2 * pybamm.arctan(1/(sto**2) - 1))
        - 2.5304 * pybamm.cos(3 * pybamm.arctan(1/(sto**2) - 1))
        + 1.2717 * pybamm.cos(4 * pybamm.arctan(1/(sto**2) - 1))
        - 0.4884 * pybamm.cos(5 * pybamm.arctan(1/(sto**2) - 1))
        + 0.1301 * pybamm.cos(6 * pybamm.arctan(1/(sto**2) - 1))
        - 0.1086 * pybamm.cos(7 * pybamm.arctan(1/(sto**2) - 1))
    )
    return dEdT

def negative_exchange_current_density(c_e, c_s_surf, c_s_max, T):
    cl_0 = pybamm.Parameter("Initial concentration in electrolyte [mol.m-3]")
    k = 6
    k_ref = k * 2 / (c_s_max * cl_0 ** 0.5)
    E_ac = 20e3
    arrhenius = np.exp(E_ac / pybamm.constants.R * (1 / 298.15 - 1 / T))
    i0 = k_ref * arrhenius * c_e ** 0.5 * c_s_surf ** 0.5 * (c_s_max - c_s_surf) ** 0.5
    return i0

def negative_diffusivity(sto, T):
    D_ref = 6e-14
    E_Ds = 3.5e4    
    arrhenius = np.exp(E_Ds / pybamm.constants.R * (1 / 298.15 - 1 / T))
    Ds = D_ref * arrhenius
    return Ds

def positive_ocp(sto):
    u_eq = (        
        3.4323
        - 0.8428 * np.exp(-80.2493 * (1 - sto) ** 1.3198)
        - 3.2474e-6 * np.exp(20.2645 * (1 - sto) ** 3.8003)
        + 3.2482e-6 * np.exp(20.2646 * (1 - sto) ** 3.7995)
    )
    return u_eq

def positive_OCP_entropic_change(sto, c_s_max):
    dEdT = (
        -0.3538 * sto ** 8
        + 1.3902 * sto ** 7
        - 2.2585 * sto ** 6
        + 1.9635 * sto ** 5
        - 0.9872 * sto ** 4
        + 0.2886 * sto ** 3
        - 0.0463 * sto ** 2
        + 0.0032 * sto
        - 1.9186e-5
    )
    return dEdT

def positive_exchange_current_density(c_e, c_s_surf, c_s_max, T):
    cl_0 = pybamm.Parameter("Initial concentration in electrolyte [mol.m-3]")
    k = 2
    k_ref = k * 2 / (c_s_max * cl_0 ** 0.5)
    E_ac = 53e3
    arrhenius = np.exp(E_ac / pybamm.constants.R * (1 / 298.15 - 1 / T))
    i0 = k_ref * arrhenius * c_e ** 0.5 * c_s_surf ** 0.5 * (c_s_max - c_s_surf) ** 0.5
    return i0

def positive_diffusivity(sto, T):
    D_ref = 4e-13
    E_Ds = 2e4    
    arrhenius = np.exp(E_Ds / pybamm.constants.R * (1 / 298.15 - 1 / T))
    Ds = D_ref * arrhenius
    return Ds

def electrolyte_diffusivity(c_e, T):
    D_c_e = (
        5.2969e-20 * c_e ** 3
        - 2.7984e-16 * c_e ** 2
        + 2.8086e-13 * c_e
        + 3.4857e-10
    )
    E_D_c_e = 16500
    arrhenius = np.exp(E_D_c_e / pybamm.constants.R * (1 / 298.15 - 1 / T))
    Dl = D_c_e * arrhenius
    return Dl

def electrolyte_conductivity(c_e, T):
    sigma_e = (
        3.2948e-10 * c_e ** 3
        - 1.6463e-6 * c_e ** 2
        + 2.2550e-3 * c_e
        + 2.5977e-2
    )
    E_sigma_e = 4000
    arrhenius = np.exp(E_sigma_e / pybamm.constants.R * (1 / 298.15 - 1 / T))
    sigmal = sigma_e * arrhenius
    return sigmal

def cation_transference_number(c_e, T):
    t_plus_ref = (
        5.7052e-17 * c_e ** 5
        - 4.7979e-13 * c_e ** 4
        + 1.2203e-9 * c_e ** 3
        - 1.1791e-6 * c_e ** 2
        + 2.8155e-4 * c_e
        + 3.5182e-1
    )
    E_t_plus = 16500
    arrhenius = np.exp(E_t_plus / pybamm.constants.R * (1 / 298.15 - 1 / T))
    t_plus =  t_plus_ref * arrhenius
    return t_plus

def thermodynamic_factor(c_e, T):
    dlnfdlnc_ref = (
        -4.4566e-10 * c_e ** 3
        + 1.5644e-6 * c_e ** 2
        - 7.4218e-5 * c_e
        - 3.1908e-2
    )
    E_dlnfdlnc = 16500
    arrhenius = np.exp(E_dlnfdlnc / pybamm.constants.R * (1 / 298.15 - 1 / T))
    dlnfdlnc =  dlnfdlnc_ref * arrhenius
    return dlnfdlnc

parameter_values["Negative electrode OCP [V]"] = negative_ocp
parameter_values["Negative electrode OCP entropic change [V.K-1]"] = negative_OCP_entropic_change
parameter_values["Negative electrode exchange-current density [A.m-2]"] = negative_exchange_current_density
parameter_values["Negative electrode diffusivity [m2.s-1]"] = negative_diffusivity
parameter_values["Positive electrode OCP [V]"] = positive_ocp
parameter_values["Positive electrode OCP entropic change [V.K-1]"] = positive_OCP_entropic_change
parameter_values["Positive electrode exchange-current density [A.m-2]"] = positive_exchange_current_density
parameter_values["Positive electrode diffusivity [m2.s-1]"] = positive_diffusivity
parameter_values["Electrolyte conductivity [S.m-1]"] = electrolyte_conductivity
parameter_values["Electrolyte diffusivity [m2.s-1]"] = electrolyte_diffusivity
parameter_values["Cation transference number"] = cation_transference_number
parameter_values["Thermodynamic factor"] = thermodynamic_factor

parameter_values.set_initial_stoichiometries(0)
pybamm.set_logging_level("NOTICE")

cycle_number = 10
experiment = pybamm.Experiment(["Rest for 30 min"])

safe_solver = pybamm.CasadiSolver(return_solution_if_failed_early=True)
simulation = pybamm.Simulation(model, parameter_values = parameter_values, experiment = experiment, var_pts = var_pts)

solution = simulation.solve(safe_solver)

### Relevant log output

```shell
Trying the code in Reproduce, I've got the error "The residual function failed at the first call. ". Searching in google, no same error occur. please help, thank you so much
```

kratman, 2024-02-02:
Hi @flag2fish Solver errors happen when the simulations are not converging. It could be that there is a problem with your custom OCP functions, it could be the ```var_pts``` grid you are using, or any number of issues. For instance, on MacOS the SEI with cracks is sensitive to the grid.

It could be difficult to debug your particular issue since you are setting the grid, using a custom parameter set, using thermal models, and using degradation models all at once. You will probably need to play around with the settings a bit to narrow down what is happening. For instance, if you use your settings with an existing parameter set, do you see the same issues?

flag2fish, 2024-02-05:
> Hi @flag2fish Solver errors happen when the simulations are not converging. It could be that there is a problem with your custom OCP functions, it could be the `var_pts` grid you are using, or any [...]

thank you for your reply. I try to add position ocp function only.
positive ocp function:
def positive_ocp(sto):
    u_eq = (        
        3.4323
        - 0.8428 * np.exp(-80.2493 * (1 - sto) ** 1.3198)
        - 3.2474e-6 * np.exp(20.2645 * (1 - sto) ** 3.8003)
        + 3.2482e-6 * np.exp(20.2646 * (1 - sto) ** 3.7995)
    )
    return u_eq
and I need to change two parameter:

parameter_values["Upper voltage cut-off [V]"] = 3.65
parameter_values["Open-circuit voltage at 100% SOC [V]"] = 3.65

and then it worked.

closed 2024-02-05 by flag2fish