pybamm-team/PyBaMM #3834: generalise simulation

martinjrobins, 2024-02-23:
### Description

at the moment simulation is a specific to the battery models and experiment class within pybamm. I'd like to propose a simpler, more general purpose mechanism so that a sequence of pybamm models can be solved (in sequence), with the next model initialised from the previous (including sensitivities). This doesn't have to be anything fancy, just a way to reinitialise a model from the solution of another model (although you could also have a base simulation class that works for a general pybamm model). I'd imagine that the simulation class could use this to simplify some of its logic as well

### Motivation

This was motivated by https://github.com/pybamm-team/PyBaMM/discussions/3804
and https://github.com/pybamm-team/PyBaMM/issues/3176

brosaplanella, 2024-02-26:
Agree, that's a great idea! I think a list of models, if possible, might be easier to start with, as it would be hard to abstract what a generic Experiment is without other examples. At the moment, it is just different models ran sequentially with certain events that dictate on when to finish a simulation, so it could all be encapsulated in a list of models. We can always think how to generalise an experiment or to write other `Step`.

tinosulzer, 2024-02-26:
Passing a list of models for the `Simulation` object might be confusing though, in that case I would suggest an entirely different class or a separate API like `Simulation.from_list` to load using a list of models instead of one model + an experiment

martinjrobins, 2024-02-26:
I think you need more than a list of models, you need:
- a list of (discretised) models
- a finish condition for model `a` (expression that has a zero crossing when the model finishes?, or list of events?)
- a map between state vector for model `a` and the state vector for model `b` (another expression tree?)
- anything else?

I don't know how you would do adjoints, but we don't do this atm anyway. We only do forward sensitivities

tinosulzer, 2024-02-26:
> a map between state vector for model a and the state vector for model b (another expression tree?)

This is currently handled by `model.set_initial_conditions_from`, which uses the `Variable.name` and the name of the variable in the `variables` list to match up the variables

How different would the models in the list be? Do you have a particular use case in mind?

martinjrobins, 2024-02-27:
> This is currently handled by `model.set_initial_conditions_from`, which uses the `Variable.name` and the name of the variable in the `variables` list to match up the variables
>
> How different would [...]

nope, this sounds perfect, happy to use this

tinosulzer, 2024-02-27:
It sounds like a more generic `Experiment` / `Step` might be the way to go then, with the `Experiment` / `Step` handling both how each step runs and when it terminates. After #3530 is merged in, the `Step` class will be responsible for modifying the `model`. This can easily be generalized beyond battery models, as long as each model being subsequently solved is not too different. 

I'm not sure we need to support the use case of solving completely different models as a top-level API (it's always possible to hack it together using `model.set_initial_conditions_from` in a for loop).

PR #4415 (2024-09-13): Support sensitivities for Experiments
# Description

Support sensitivity calculation for `pybamm.Simulation`, including experiments

Fixes #3834 

Note that although the sensitivity calculation for `pybamm.ScipySolver` appears to work, it seems like there are bugs in this solver for experiments (see #4429) so this is currently not tested.

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breakin [... cut]
closed 2024-09-13 by martinjrobins