pybamm-team/PyBaMM #3836: [Bug]: `import pybamm` is breaking

arjxn-py, 2024-02-24:
### PyBaMM Version

24.1

### Python Version

3.8-3.12

### Describe the bug

There might be some instance in #3578 where non required dependency has been imported by default, which is causing failure, [related build logs](https://github.com/arjxn-py/PyBaMM/actions/runs/8032837736/job/21942676799). 

### Steps to Reproduce

* try installing latest built wheels on develop
* do `import pybamm`

agriyakhetarpal, 2024-02-25:
Note: this happens with `sympy` as well

agriyakhetarpal, 2024-02-25:
Also, it's weird that the tests for `have_optional_dependency` in `test_util.py` didn't catch `typing-extensions` and `sympy` – is this because we are not mocking all of the optional dependencies inside that test case? IMO, we should mock all of the optional dependencies to ensure that this problem doesn't get repeated in the future.

arjxn-py, 2024-02-26:
I tried importing `typing-extensions` as an optional dependency but it is being imported outside a function in `unit/type_definitions.py` and then those types in `type_definitions.py` are then being imported and used as args at several instances, which is making `have_optional_dependency` unable to resolve the issue.

arjxn-py, 2024-02-26:
How about we consider making `typing-extensions` a required dependency?

PR #3848 (2024-03-05): Make `typing-extensions` and `sympy` required dependencies to fix PyBaMM import
# Description

Fix broken import of PyBaMM raised by type hinting.

Fixes #3836 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which f [... cut]
closed 2024-03-05 by kratman