pybamm-team/PyBaMM #3839: Technical roadmap

brosaplanella, 2024-02-26:
In our last developer meeting we agreed to hijack the next one (18th of March 4pm UK time) to discuss the technical roadmap for PyBaMM.

The motivation is that for the past few months we have been focusing mostly on infrastructure and smaller issues, and there are some major projects to undertake that would benefit of a more structured vision (e.g. refactoring). The goal of this upcoming meeting is to discuss what this priorities should be and choose a coordinator/lead for each one. Then, the various people/teams interested in working on each priority can discuss and work on the arising issues.

This issue is to collect any thoughts regarding the technical roadmap both before and after the meeting.

tinosulzer, 2024-02-27:
## Model entry points

We don't have as many contributors adding models to PyBaMM as we'd like. There are a few reasons for this
- PyBaMM’s submodel structure is too complex, huge barrier to entry
- You have to add things to the PyBaMM repo, uncertain ownership

We propose "model entry points" as a solution to this. Using entry points, contributors can create models in their own repositories and make them easily available to the rest of the community, without the models having to be added to PyBaMM itself. This will enable a community for battery models that use the PyBaMM infrastructure but are not owned by the PyBaMM team.

### Advantages
- Contributors get to keep models in their own repositories, retain ownership, choose license terms
- PyBaMM team doesn’t have to endorse models we don’t like
- All of GitHub’s functionality (issues, discussions, pull requests) are available for each model

### How to start

- Set up a separate repository with an example of adding a model separately outside of PyBaMM
- Make required changes to PyBaMM to enable this
    - Entry points for models
    - Making basic models compatible with experiments
    - Figure out how to deal with new parameters (define parameters inside the model itself?)
- Make cookiecutter repository to help with setting up, adding entry points, publishing to pypi, tests, docs, specifying pybamm version(s), etc (as part of the GSoC coookie-cutter project)

tinosulzer, 2024-02-27:
## Parameter refactor

Refactor parameter values to use the BPX format by default.
Using nested dictionaries allows cleaner definition of negative / positive and also different components of the battery.
This is a big breaking change so we should add a conversion class from legacy parameter sets to new ones, and legacy parameter sets should still work for a while.

tinosulzer, 2024-02-27:
## 3D simulations

@tlroy has been working on some 3D electrochemical models in firedrake (a python-based finite element code similar to FEniCS). This provides a pathway to allow full 3D simulations in PyBaMM. Very early stage / exploratory at this point

TomTranter, 2024-03-01:
I'd be interested in 3d models and it aligns with what I've done on jellyrolls

brosaplanella, 2024-03-15:
## Tidy up sumbodel structure
The current submodel structure is getting a bit messy, especially with degradation models. Allowing for model entry points (https://github.com/pybamm-team/PyBaMM/issues/3839#issuecomment-1966614301) will certainly facilitate the code. We should also enable (or publicly advertise if already possible) a way to assemble models as a dictionary of submodels and reduce the number of options to the most standard models (leaving edge cases to be defined as dictionaries).

In my opinion, the new structure should be designed generic enough to allow new chemistries (e.g. solid state electrolytes, lithium-sulfur...) if we wish to include them in the future.

martinjrobins, 2024-03-15:
> ## Refactor backend to be model agnostic
As well as processed variables I think the Experiment class should have a base class that is not battery specific (see https://github.com/pybamm-team/PyBaMM/issues/3834). Same with Simulation class probably...

martinjrobins, 2024-03-15:
> ## Parameter refactor
> Refactor parameter values to use the BPX format by default. 

Isn't the BPX format for DFN models only? I would have thought that the set of parameters that pybamm uses is much broader than what BPX defines. Then there is non-battery models as well, equivilent circuit models etc.

martinjrobins, 2024-03-15:
# Parameter Inference

I'd like to continue making pybamm easier to work with for the parameter inference libraries like PyBOP (https://github.com/pybop-team/PyBOP). At the moment I think this includes:
- GPU as well as multithreaded support for running many simulations in parallel (https://github.com/pybamm-team/PyBaMM/issues/3826)
- Integrate sensitivities more into Simulation (https://github.com/pybamm-team/PyBaMM/issues/3834) 
- Run multiple simulations from different starting points (https://github.com/pybamm-team/PyBaMM/issues/3713). Should be possible to reset the starting points during the middle of a solve
- others?....

brosaplanella, 2024-03-20:
# Summary of roadmap meeting 18/03/24
The main outcomes of the meeting were that we will initially work on the following areas:
- Model refactoring (aka Models): This is basically the [Model entry points](https://github.com/pybamm-team/PyBaMM/issues/3839#issuecomment-1966614301) suggestion above, but also incorporates the [Refactor backend to be model agnostic](https://github.com/pybamm-team/PyBaMM/issues/3839#issuecomment-1999415696) suggestion. This effort will be led by @tinosulzer.
- Parameter refactoring (aka Parameters): This is the [Parameter refactor](https://github.com/pybamm-team/PyBaMM/issues/3839#issuecomment-1966635559) suggestion above. This effort will be led by @kratman and @BradyPlanden.
- Solver improvements (aka Solvers): This is the [Parameter inference](https://github.com/pybamm-team/PyBaMM/issues/3839#issuecomment-1999464816) suggestion above, renamed to avoid confusion with the other project. This effort will be led by @martinjrobins.

This does not mean we are dropping the other areas above, just leaving them for a later stage. 

I will now create separate issues for each area so people can continue the discussions there and organise themselves. I will keep this issue open to collect other suggestions that arise.