pybamm-team/PyBaMM #3844: Restructure "Creating Models" notebooks

brosaplanella, 2024-02-28:
After reviewing the "Getting Started" notebooks, the next step is to restructure the "Creating Models" ones. The idea is to have a cohesive set of notebooks that build progressively more complex models up to DFN (potentially with degradation). The inspiration is the ["12 steps to Navier Stokes" tutorial](https://lorenabarba.com/blog/cfd-python-12-steps-to-navier-stokes/).

We will not have 12 steps, at least initially, but so far the structure would be:
1. Reservoir model (ODEs)
2. SPM (parabolic PDEs)
3. SPMe (parabolic PDEs in multiple domains, multiscale formulation appears)
4. DFN (parabolic & elliptic PDEs in multiple domains)
5. DFN + degradation (maybe SEI? Includes all of the above, but more complicated)

This is subject to change, as the notebooks are being edited/written (e.g. some might need to be split into two).

This issue would go hand-in-hand with this: https://github.com/pybamm-team/PyBaMM/issues/3839#issuecomment-1966614301

tinosulzer, 2024-02-28:
Do we want all the models to be battery models or should we start with generic models? Pro of battery models is it's more interesting, con is it requires loading in a lot of parameters

If we can have generic models, before 1 could be solving an exponential decay equation (introduce the concept of a variable, rhs, initial conditions, etc) and between 1 and 2 could be diffusion in a cartesian domain.

Other comments:
- Full SPMe is tricky because it requires a complex expression for the voltage
- It might be helpful to include some intermediate models between SPMe and DFN that introduce the coupling of potential as an algebraic equation

tinosulzer, 2024-02-28:
Here are some models I wrote a while back, they don't work any more but might be useful intermediates to explain some of the model structure
```python
import pybamm

class BaseModel(pybamm.lithium_ion.BaseModel):
    def __init__(self, name):
        super().__init__({}, name)
        self.variables = {
            "Time": pybamm.t,
            "x": pybamm.standard_spatial_vars.x,
            "x_n": pybamm.standard_spatial_vars.x_n,
            "x_p": pybamm.standard_spatial_vars.x_p,
            "r_n": pybamm.SpatialVariable("r_n", domain=["negative particle"]),
            "r_p": pybamm.SpatialVariable("r_p", domain=["positive particle"]),
        }

    def new_empty_copy(self):
        return pybamm.BaseModel.new_empty_copy(self)

def spm():
    model = pybamm.lithium_ion.SPM(name="SPM")
    model.variables.update(
        {
            "r_n": pybamm.SpatialVariable("r_n", domain=["negative particle"]),
            "r_p": pybamm.SpatialVariable("r_p", domain=["positive particle"]),
        }
    )
    variables = [
        "X-averaged negative particle concentration",
        "X-averaged positive particle concentration",
    ]
    return model, variables

def spm_no_r():
    model = pybamm.lithium_ion.SPM({"particle": "uniform profile"}, name="SPM_no_r")
    variables = [
        "Average negative particle concentration",
        "Average positive particle concentration",
    ]
    return model, variables

def spme():
    model = pybamm.lithium_ion.SPMe(name="SPMe")
    model.variables.update(
        {
            "r_n": pybamm.SpatialVariable("r_n", domain=["negative particle"]),
            "r_p": pybamm.SpatialVariable("r_p", domain=["positive particle"]),
        }
    )
    variables = [
        "X-averaged negative particle concentration",
        "X-averaged positive particle concentration",
        "Electrolyte concentration",
    ]
    return model, variables

def dfn_no_r():
    model = pybamm.lithium_ion.DFN({"particle": "uniform profile"}, name="DFN_no_r")
    variables = [
        "R-averaged negative particle concentration",
        "R-averaged positive particle concentration",
        "Electrolyte concentration",
        "Electrolyte potential",
        "Negative electrode potential",
        "Positive electrode potential",
    ]
    return model, variables

def dfn():
    model = pybamm.lithium_ion.DFN(name="DFN")
[... 148 lines cut ...]
                    "Electrolyte potential": phi_e,
                    "Negative electrode potential": phi_s_n,
                    "Positive electrode potential": phi_s_p,
                    "Interfacial current density": a_j,
                }
            )

    model = Model()
    variables = list(model.variables.keys())
    return model, variables

def solve_plot(model, variables):
    sim = pybamm.Simulation(model)
    sim.solve([0, 3600])
    sim.plot(variables)

# solve_plot(*spm())
# solve_plot(*spm_no_r())
# solve_plot(*reduced_c())
# solve_plot(*spme())
# solve_plot(*reduced_c_phi())
# solve_plot(*reduced_c_phi_j())
# solve_plot(*dfn_no_r())
# solve_plot(*dfn())
```