pybamm-team/PyBaMM #3847: [Bug]: Fix `Lychee Broken Link Checker` for Stack Overflow Links

cringeyburger, 2024-02-28:
### PyBaMM Version

Develop

### Python Version

3.11.5

### Describe the bug

PyBaMM's Lychee link checker is unable to get through Stack Overflow's security to verify links. 

### Steps to Reproduce

By using the `Check URLs using Lychee` GitHub Action of PyBaMM

cringeyburger, 2024-02-28:
Things I have tried:
1. Setup a `user-agent` of different platforms -> No change in result
2. Try using Wayback Machine or `web.archive.org` to replace Stack Overflow links -> works completely fine

Hence I want to propose that we can use the Internet Archive links to replace Stack Overflow links.

agriyakhetarpal, 2024-02-28:
I am slightly against the removal of Stack Overflow links (they help with knowledge transfer and attribution). However, wherever applicable, we can grab short permalinks for any of the longer links, and then ignore Stack Overflow links altogether – permalinks won't fail link checkers, even if the platform is getting aggressive towards them.

I have just submitted https://github.com/lycheeverse/lychee/issues/1384 to request functionality to ignore particular sections of files, let's hope they find it useful and we can then ignore the all-contributors section in the README.md file. As pointed by @kratman, I am indeed favour of removing excess links as a long term solution.

kratman, 2024-02-28:
> I have just submitted [lycheeverse/lychee#1384](https://github.com/lycheeverse/lychee/issues/1384) to request functionality to ignore particular sections of files

I am not sure we need this for the all contributors. We can just add a markdown link to an `all_contributors.md` that shows everything. Then we just ignore the stand alone file

agriyakhetarpal, 2024-02-29:
> I am not sure we need this for the all contributors. We can just add a markdown link to an all_contributors.md that shows everything. Then we just ignore the stand alone file

That would be great too, it would simplify a lot of things. In other words, we received a workaround at https://github.com/lycheeverse/lychee/issues/1384#issuecomment-1970108569, which is pretty understandable and can be used with GitHub Actions – it's just a bunch of shell operations and pipes. So either of these should work.

<hr>

Note to self: this might break the https://pybamm.org/teams/ page, I will have to fix that later

kratman, 2024-02-29:
I can take this up this weekend. The failing check annoys me, so I will do the fixes

tinosulzer, 2024-03-05:
Looks like it's fixed now

kratman, 2024-03-05:
@tinosulzer I still want to make some changes since that check fails a lot. I want it a bit more robust

tinosulzer, 2024-03-05:
Ok

PR #3894 (2024-03-14): Move all contributors to a different file
# Description

Moves the all contributors section out of the README.md file so that it can be ignored by Lychee.

Fixes #3847

## Type of change

Minor fix to make the CI more stable. Since the all contributors section contains links to contributors personal web profiles, they can cause lychee to fail if a users site is not working. Issues like this will become more common as we get more contributors.

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (or `$ nox -s pre-commit`) (see [C [... cut]
closed 2024-03-14 by kratman