pybamm-team/PyBaMM #3856: Minimum voltage [V]

Santiagopeacely, 2024-03-02:
Helllo，
**Lower voltage cut-off [V]': 2.5**  of ParameterValues("Chen2020"). But I can discharge under 1C for 1 hour, the voltage can drop below 2.5 V, indicating that the cut-off voltage here did not work, when I adjusted 1 hour to 2 hours, the results prompted me **Experiment is infeasible: 'event: Minimum voltage [V]' was triggered during 'Discharge at 1C for 2 hour'. The returned solution only contains up to step 1 of cycle 1.** Where should I look for **Minimum voltage [V]**?
This is my code
**_import pybamm
experiment = pybamm.step.string("Discharge at 1C for 1 hour", period = "0.1 minute", temperature = "25oC", tags = ["tag1"])
parameter_values = pybamm.ParameterValues("Chen2020")
model = pybamm.lithium_ion.DFN()
sim = pybamm.Simulation(model, parameter_values = parameter_values, experiment = experiment)
sim.solve()
sim.plot()_**
[image]

rtimms, 2024-03-02:
You can put the voltage cut off in the experiment, e.g.“Discharge at 1C for 1 hour or until 2.5V”. 

To find the minimum voltage just look for that key in the parameter dictionary or do a search, e.g. `parameter_values.search(“voltage”)`.

Santiagopeacely, 2024-03-02:
Thank you for your answer. I know Experiment can complete this operation, but I want to use the parameters inside to automatically stop. If the discharge does not stop when the cut-off voltage or 0% SOC is reached, what is the significance of this parameter? I couldn't find the minimum voltage through searching, it's not in the parameter table of "chen2020". Based on operation, it's probably around 1.5V.

Santiagopeacely, 2024-03-02:
When I try to change the parameters in the table，parameter_values["Open-circuit voltage at 100% SOC [V]"] = 
parameter_values["Open-circuit voltage at 0% SOC [V]"] = ，When changing the cut-off voltage of 100% SOC, there will be a change in the output, while when changing the 0% SOC, there will be no change in the output, regardless of how it is changed, even if it is 0. So I hope you can give me a reason. 
Thank you so much

tinosulzer, 2024-03-05:
The parameter is "Lower cut-off voltage [V]". Use that parameter to explicitly define the minimum voltage in your experiment based on the parameter set.

Santiagopeacely, 2024-03-07:
Thank you very much for your answer！

closed 2024-03-07 by tinosulzer