pybamm-team/PyBaMM #3872: Evaluate Experiment at specified time interval

mmichal9, 2024-03-07:
### Description

Hello there! 

I am current working on a use case where I would like to simulate the battery charge / discharge cycle at a given time and access the voltage current data in "real time" rather than let full simulation complete and the access the data at the end. I would also like to also control next steps based on the outputs (voltage / current). 

In short, I would like to have my own control system deciding next steps (cc / cv charge discharge) based on the current state of the simulation. 
Is this possible to do ? 

So far I have looked at passing an additional parameter to each experiment to force it to run on a small delta t and repeat until voltage is met this works ish ok, but it is really slow... I suspect due to redefining sim every time.
Example code : 
```
    def _solve(self, experiment):
        self.sim = pybamm.Simulation(self.model, experiment=experiment, parameter_values=self.params)

        if self.sol is None:
            self.sol = self.sim.solve()
            return
        self.sol = self.sim.solve(starting_solution=self.sol)

    def cc_dt(self, current: float, voltage_limit: float, charge: bool = True, t_stop=0.1) -> bool:
        direction = "Charge"
        if not charge:
            direction = "Discharge"

        experiment = pybamm.step.string(f"{direction} at {current} A for {t_stop} seconds or until {voltage_limit} V", period = PERIOD)
        self._solve(experiment=experiment)

        print("t:", self.get_present_time(), "v:", self.get_present_voltage())
        if self.get_present_voltage() >= voltage_limit - DELTA:
            return False

        self._current_cycle +=1
        return True
```

The other option i have seen is to pass t_eval to a model. Similar to what has been described in the following [notebook](https://docs.pybamm.org/en/latest/source/examples/notebooks/models/SPM.html). This is faster, however In this case i have issue with not being able to define the experiments explicitly and the current / voltage values seen to be the same no matter the step. 

What would be the best way to control the experiments dynamically based on the current state of the simulation? Is there a way to access the data of the simulation when it is running and direct it to execute different steps ? 

### Motivation

Dynamic control of the battery experiment during simulation

tinosulzer, 2024-03-07:
Dynamic simulation control is outside of scope and not our priority at the moment. I think the way you're trying is about how I would do it. You are correct that this is slow due to overheads from restarting the simulation, and also the adaptive solvers perform best when they can take large time steps (not limited to 0.1s). 

We're working on reducing the simulation overheads, and refactoring how experiments are done, but this kind of use case is not what we cater for. You could use PyBaMM to create and discretize the model then write your own solver for this use case.

mmichal9, 2024-03-07:
I suspected this might be the case since this seem a bit outside of the general use case for what I saw in pybamm so far. 

Regarding custom solvers I have started looking into the internals of the base_solver.py since I don't need anything too fancy at this stage. Going thought the code, I have a bit of the hard time understanding exactly where I could start. Do you have any recommendations ?

mmichal9, 2024-03-07:
Alternatively, I was also considering running full step (charge) and then streaming the data to some custom controller class. if the controller decided to run a new step (rest) I would pass the state of the simulation at that time. However I did not find a way to save the state of the simulation at specific times when looking at the callbacks. Is there a better way of doing it? Or would you suggest looking into the custom solver option would be a better option?

tinosulzer, 2024-03-07:
Take a look at `scipy_solver.py`

kratman, 2024-03-14:
@mmichal9 Do you need any other help with this ticket?