pybamm-team/PyBaMM #3881: How to obtain the radial stress distribution of particles in the swelling model

Santiagopeacely, 2024-03-11:
[image]
Hello,Sir
  I can only see the stress distribution along the particle surface in the variables of this model. I can not find one variable of the stress distribution inside the particle. Can this model obtain radial stress instead of surface radial stress?
  Looking forward to your answer, thank you very much!

rtimms, 2024-03-11:
cc @weilongai

weilongai, 2024-03-11:
I'm afraid there are no built-in variables for the radial stress inside particles in PyBaMM, but they can be easily calculated from lithium concentrations. See Eqs. (22-23) in [Journal of The Electrochemical Society, 154(10), A910-A916, 2007](https://iopscience.iop.org/article/10.1149/1.2759840)

Santiagopeacely, 2024-03-11:
Thank you very much for your clarification. Wishing you success in your research and happiness every day！

L Y?
***@***.***

抄送: "L ***@***.******@***.***&gt;;

kratman, 2024-03-14:
@Santiagopeacely It looks like you got an answer for this ticket, so I am going to close this for now. If there is anything more feel free to reopen the ticket.

closed 2024-03-14 by kratman