pybamm-team/PyBaMM #3898: Update source installation instructions for linux to include more distro-specific instructions

santacodes, 2024-03-17:
Currently the install from source and other areas of installation related references in the documentation lack linux distribution specific steps. Linux distributions can have different package managers and therefore different package names, so to generalize them under `GNU/Linux` wouldn't be specific as it references the linux base kernel.
Instead we could add the following distribution specific steps as they are the most commonly used among developers and personal users.

- Ubuntu/Debian (apt)
- RHEL/Fedora (yum/rpm/dnf)
- Arch Linux (pacman)
- NixOS (nix)
- Windows (chocolatey) if applicable

Most of the other distributions are mostly built around these base distributions, so adding these package managers to the documentation would also cover distributions which use the mentioned package managers.

agriyakhetarpal, 2024-03-17:
Thanks, @santacodes – incorporating more instructions to better match other Linux distributions would certainly be useful for the "Installation from sources" page. I would leave NixOS out for the time being (or make that section minimal instead).

For Windows, we do mention that the IDAKLU solver is not available when building from source – however, the truth is that the compilation is a bit tricky and it can take time to build. Some general instructions there would suffice (we don't use Chocolatey as of now – instructions for MSVC would be preferred), but can be taken up separately as well.

If you're interested in working on this issue, please let me know and I can assign you to it.

Feel free to use the available Sphinx extensions to organise the content (i.e., through tabs, admonitions, dropdowns, and so on) as you like it.

santacodes, 2024-03-17:
Sure I'd love to take this issue and work on it

PR #3914 (2024-03-23): Added more Linux distribution specific installation procedures
# Description

Install from source and other prerequisites dependency installation procedures lacked distribution specific steps. Common Linux package managers (`apt`, `dnf`, `pacman`) specific command tabs are added to the documentation.

Fixes #3898 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New fea [... cut]
closed 2024-03-23 by agriyakhetarpal