pybamm-team/PyBaMM #3909: [Roadmap area] Parameter refactor

brosaplanella, 2024-03-20:
Refactor parameter values to use the BPX format by default.
Using nested dictionaries allows cleaner definition of negative / positive and also different components of the battery.
This is a big breaking change so we should add a conversion class from legacy parameter sets to new ones, and legacy parameter sets should still work for a while.

_Originally posted by @tinosulzer in https://github.com/pybamm-team/PyBaMM/issues/3839#issuecomment-1966635559_

rtimms, 2024-03-20:
FYI we (Ionworks) have a FUSE internship open for applications that is related to this, so we can support this over the summer

tinosulzer, 2024-03-22:
Also, when we do this, it would be great to do a deep review of existing parameter sets, including removing parameters that are not provided in the original reference (e.g SEI parameters for a performance-only model). See https://github.com/pybamm-team/PyBaMM/pull/3919#pullrequestreview-1954878606

brosaplanella, 2025-07-08:
Following https://github.com/pybamm-team/PyBaMM/issues/5064 this area is now paused until further notice, as it is no longer a priority.

closed 2025-07-08 by brosaplanella